OctoPrint/plugins.octoprint.org #94: Anywhere = Spyware

DeonEksteen, 2017-07-03:
READ FIRST, THEN DELETE:

What kinds of issues you should report here:

  * missing or broken homepage, source or download URLs
  * broken images
  * incorrect compatibility information
  * ... in short, anything related to the data on plugins.octoprint.org

What you should not report here:

  * Any kinds of issues with the plugins themselves - report those
    to the plugin author instead!

Thanks :)

foosel, 2017-07-04:
What link? URL please? The link in the plugin repository points to the plugin's Github page.

@kennethjiang any idea what OP might be talking about?

kennethjiang, 2017-07-04:
Hmmm I don't have too much  context on what @DeonEksteen  talked about either. My plugin uses auth0 for authentication from which I received nothing but email as identifier. Anyone who doesn't feel comfortable with giving out the email address can deny so on auth0 signup page, which clearly states the authorization terms. 

auth0 is a reputable service and the way I use it is completely in compliance with their user policy and general web app practices.

@DeonEksteen please shed some lights on your claims. I have spent a lot of effort developing free octoPrint plugins without any returns except the fact that many 3D printer users have slightly easier lives because of my plugins. The last thing I want to hear is some speculations on my intention, or the intension of any software developer who is devoted to building open source softwares.

kennethjiang, 2017-07-04:
Another user pointed out that there is a link from my plugin document to the website of anywhere.io, which contains nothing but signup form.

anywhere.io is a stealth mode startup that my friends built who let me use for free. I will remove its link from my doc if that's causing a concern.

foosel, 2017-07-04:
#95 is merged which removed that link. If that's the link @DeonEksteen took offense in. 

In general my two cents on this: It's a plugin that allows to monitor your instance from outside your own network. It's obligatory that it needs some cloud counterpart for that, it simply is not possibly to solve this differently with current technology. In this case that cloud counterpart is that Anywhere platform, which also is reflected in the name. If someone is not comfortable sharing data with that cloud platform, that's their right. Nobody forces anyone here, and that plugin then is not the right for them. But it's perfectly valid to have this plugin in the repository (and imho also for it to link to the cloud platform in question).

DeonEksteen, 2017-07-04:
Hi guys
I do appreciate your concerns with the issue I experienced and would like to inform you what happened.  
http://prntscr.com/frh1hw
Where the arrow is was a link highlighted in blue to "Anywhere" plugin. I see this has now been removed. When I clicked on that link it gave me the option to connect my devices and asked for a Username, Telephone number and E-Mail address. I was then taken to a page to create a survey form. When I clicked the save button the survey was created but it wanted me to repeat the process again and again. Nowhere could I connect any 3d printer or any other devices at that stage. I really cannot say what happened at that exact moment when I tried to register but I needed to bring this under your intention as what I experienced contained all the characteristics of spyware. 
Thanks for all your hard work.
Regards
Deon

foosel, 2017-07-04:
Ok. Considering this closed then. That link you followed was the link to the cloud platform that this plugin is based on (which I guess was only provided for reasons of transparency), NOT the plugin's homepage or anything like that.

closed 2017-07-04 by foosel