OctoPrint/plugins.octoprint.org #100: Duplicate

ikishk, 2017-07-17:
This is a duplicate of http://plugins.octoprint.org/plugins/cost/  code is slightly diff, so not a complete clone.

foosel, 2017-07-18:
I know, and I pondered whether to merge #99 or not for that very reason. In the end, I decided for it and let the community decide on which one they prefer. I don't want to police entries too much to be honest, that just leads to conflict.

I'm sure @malnvenshorn had a valid reason to develop their own version, and I don't think it's my place to prevent their efforts from being published on the repository.

malnvenshorn, 2017-07-18:
I'm aware that a similar plugin exists. I tried the plugin from @jasiek, actually. But I wasn't happy with the fact that it requires fixed values for cost per weight and cost per time. If one parameter changes, e.g. price per kWh, you have to calculate the new value by your own. That and the fact that I wanted to look into the plugin development of OctoPrint I decided to write my own plugin from scratch.

I wasn't sure weather I should publish my plugin or not, because I wrote this plugin for myself. I decided to add it to the plugins repository and everyone can decide which one they want to use.

foosel, 2017-07-18:
Sounds like more than a valid reason for me :)

As long as we don't get *exact* duplicates (and here there's a different in functionality - the existing one allows some things the new one doesn't like configurable currency, the new one on the other hand also factors in power consumption), personally I think everything is fine. Similar but distinct plugins make sense.

foosel, 2017-07-18:
Closed because I'm not going to revert the merge. Comments are still open for discussion however.

closed 2017-07-18 by foosel