OctoPrint/plugins.octoprint.org #104: Major problem

karatetrigger, 2017-07-29:
The screenshots that he has in those pictures were of pictures of my 3d printer. I do not know the author, how he got those screenshots of my 3d printer printing, or my webcam feed, but I have a MAJOR problem with this right now.

kantlivelong, 2017-07-29:
Hello,

Could you provide some proof? (You can email to me if desired. )

kantlivelong at gmail dot com

kantlivelong, 2017-07-30:
Hi @karatetrigger,

I've not received a response from the developer of the plugin and thus have removed the images on http://plugins.octoprint.org/plugins/anywhere/.

They are not completely removed as the changes are in source control but they will no longer appear on the website.

kennethjiang, 2017-07-30:
Sorry for delay. I hadn't had connections until today.

@karatetrigger My apology for using those pics. I found those pics through an ip camera search engine. But I should have tried to find your contact and asked your permission. Now I know that you don't these pics to be used, I will replace them with my own webcam screenshots (not as good quality as yours) when I get home. Will make sure I fix that by end of day tomorrow. 

Just a friendly reminder that your webcam seems to be indexed by ip camera search engines. Do you somehow make it exposed to internet? You may want to check your network settings if it was not your intention to begin with.

kantlivelong, 2017-07-31:
Thanks @kennethjiang,

Submit your PR with the required changes and I will merge it in ASAP.

kennethjiang, 2017-07-31:
@kantlivelong just sent PR #105 .

@karatetrigger I removed all your pics from this repo. Will work on other repos to clean them up in the next couple days. My apology again!

kantlivelong, 2017-07-31:
Thanks @kennethjiang 

@karatetrigger are we good to close this?

foosel, 2017-08-22:
Closing this since the images have been replaced. If there is still a problem @karatetrigger, please get in touch here again (comments are still possible).

closed 2017-08-22 by foosel