OctoPrint/plugins.octoprint.org #141: Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

CADMunkie, 2017-12-29:
I have copied the information from the dialogue boxes

What kinds of issues you should report here:

Installing plugin "FilamentManager" from https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip...
/home/pi/oprint/bin/python -m pip install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

the error box:

 Something went wrong
Installing the plugin from "https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip" failed: Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Thanks :)

SagiRokach, 2017-12-29:
I have the same exact problem.

Any idea how to solve it?

kantlivelong, 2017-12-29:
Hello,

This tracker is for:
  * missing or broken homepage, source or download URLs
  * broken images
  * incorrect compatibility information
  * ... in short, anything related to the data on plugins.octoprint.org

If the issue only occurs when installing a specific plugin please report to the appropriate plugin repository (In this case https://github.com/malnvenshorn/OctoPrint-FilamentManager). 

If this issue affects all plugins please submit a ticket to https://github.com/foosel/OctoPrint

closed 2017-12-29 by kantlivelong