OctoPrint/plugins.octoprint.org #143: Unable to install plug in

CADMunkie, 2018-01-04:
READ FIRST, THEN DELETE:

What kinds of issues you should report here:

Unable to install plug in

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-YUftLj/psycopg2/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

What you should not report here:

Thanks :)

kantlivelong, 2018-01-04:
What plugin?

CADMunkie, 2018-01-04:
Filament Manager for OctoPrint

Thanks

kantlivelong, 2018-01-04:
Installs OK here.

This tracker is for:
  * missing or broken homepage, source or download URLs
  * broken images
  * incorrect compatibility information
  * ... in short, anything related to the data on plugins.octoprint.org

If the issue only occurs when installing a specific plugin please report to the appropriate plugin repository (In this case https://github.com/malnvenshorn/OctoPrint-FilamentManager). 

If this issue affects all plugins please submit a ticket to https://github.com/foosel/OctoPrint

closed 2018-01-04 by kantlivelong