OctoPrint/plugins.octoprint.org #144: How do I use it !!!

michael5210, 2018-01-04:
i'm a newbie and it is not clear - how to use it

kantlivelong, 2018-01-04:
This tracker is reserved for:

- missing or broken homepage, source or download URLs
broken images
- incorrect compatibility information
- ... in short, anything related to the data on plugins.octoprint.org

If this issue only applies to a specific plugin please report to the appropriate plugin repository.

If this issue affects all plugins please submit a ticket to https://github.com/foosel/OctoPrint

closed 2018-01-04 by kantlivelong

sbts, 2018-01-04:
@michael5210 as @kantlivelong indicated this is the wrong place for any sort of questions outside those listed.

However, if you look at the [Octoprint website](http://octoprint.org] you will find information about various ways and places to as questions and get assistance.