OctoPrint/plugins.octoprint.org #146: 404 Error on "More..." link

MikeZets, 2018-01-05:
On page    http://octoprint.github.io/plugins.octoprint.org/
the "More..." link goes to a missing page and displays a 404 error message.

MikeZets, 2018-01-05:
> Hi @MikeZets <https://github.com/mikezets>
>
> Can you tell me what URL linked to http://octoprint.github.io/
> plugins.octoprint.org/ ?

Sorry.  I couldn't navigate my way back to the referring URL.  I was in the
Facebook Users Group Photos section and found the link to Plug-ins.  Went
there and then got the 404 when I tried to click on "More..."

*Mike Zets*
8346 Ledgestone Avenue
Sylvania, OH  43560
*412 670-7646 (*Cell)
MikeZets@gmail.com
*www.MikeZets.info <http://www.mikezets.info/>*

foosel, 2018-01-11:
Sounds like someone linked to the public github pages endpoint instead of the actual domain. No wonder that that made things hiccup. I'll see if I can't add some kind of JS redirect...

commit 08272ee7 (2018-01-18): Redirect github.io page to plugins.octoprint.org via JS

Closes #146
closed 2018-01-18 by foosel