OctoPrint/plugins.octoprint.org #234: Project Kronos Data Collector - description is misleading

arthurlutz, 2018-09-27:
The description of https://plugins.octoprint.org/plugins/kronos_data_collector/ is misleading. 

The description starts by stating functionnality about detecting failure with neural nets, automatically pause or alert the user. But these features are not at all include in the plugin and are not yet implemented. 

I think the description should clearly state that this plugin (for now) *only* sends your data to an Amazon S3 bucket. Maybe it just a matter of putting the paragraph about sending the data first and the "future development" of the neutral net at the bottom. 

There are various issues about this plugin https://github.com/MrBreadWater/project-kronos-data-collector/issues 

This is an interesting project, but it probably needs a clarification in terms of it's  "opensource" / "commons" approach.

foosel, 2018-09-27:
Hm... I disagree.

The description first explains what Project Kronos is (as in, what is the goal of the whole thing and what *will be implemented* once there's enough data) and then goes on to explain what that specific plugin does. 

It doesn't claim that plugin is anything but a data collector, but it gives context as to why the plugin collects data.

@kantlivelong what's your take on this?

@MrBreadWater want to chime in here?

kantlivelong, 2018-09-27:
I'd have to agree with you @foosel. The plugin is pretty clear on what it does in the plugin repo, source repo, and after installation via the wizard.

MrBreadWater, 2018-09-27:
I certainly thought I was clear in what the plugin is and does. The first thing I explain is the goal of Project Kronos, and then right afterwards explains what the Data Collector plugin does. I even named the plugin accordingly, so as to differentiate it from the final project again.

If it becomes a problem again, I’ll add another disclaimer at the end.

arthurlutz, 2018-10-01:
@foosel @kantlivelong Thanks for taking a look at this issue. Feel free to close the issue if you think this is not a problem. 

@MrBreadWater Thanks for your offer of clarification.

kantlivelong, 2018-10-02:
@arthurlutz: better to bring it up than say nothing. Thanks :)

Closing

closed 2018-10-02 by kantlivelong