OctoPrint/plugins.octoprint.org #246: NOT AN ISSUE!  First thing to say. You are great for creating this!  Many thanks.  I am very impressed at how nice it is.  You know both programming and how to give the information to people!  A suggestion please.  add a way to override the plugin if desired.  I have a 15" touchscreen and I don't need the small interface :)  Also, maybe a clock on the TouchUI please?

johne3819, 2018-11-13:
READ FIRST, THEN DELETE:

What kinds of issues you should report here:

  * missing or broken homepage, source or download URLs
  * broken images
  * incorrect compatibility information
  * ... in short, anything related to the data on plugins.octoprint.org

What you should not report here:

  * Any kinds of issues with the plugins themselves - report those
    to the plugin author instead!

Thanks :)

johne3819, 2018-11-13:
I would like to tell the author but I cannot find a way to contact.  please help

jubaleth, 2018-11-13:
Upon second reading of the ticket, I think I need to drink more coffee before responding 🤣 If I understand correctly, you want to make a feature request for TouchUI, in which case BillyBlaze's repository is the right place to go =)

Cheers (with more coffee this time)