OctoPrint/plugins.octoprint.org #254: Unable to install Bed Visualizer

techpad-marcb, 2018-12-02:
**Hardware:**
Orangepi Zero 512mb

**Software:**
Armbian_5.59_Orangepizero_Ubuntu_bionic_next_4.14.65
OctoPrint 1.3.9

**Issue:**
When installing this plugin I it seems to just keep going and never times out or finishes. Several tries now on several versions of OctoPrint and still have the same issues.

Numpy seems to not be installing for some reason.
URL: https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip

**Install Log:**
Installing plugin from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip...
/home/octoprint/OctoPrint/venv/bin/python -m pip install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip --no-cache-dir
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
Downloading https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
Requirement already satisfied: OctoPrint in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg (from Bed-Visualizer==0.1.7) (1.3.9)
Collecting numpy (from Bed-Visualizer==0.1.7)
Downloading https://files.pythonhosted.org/packages/2d/80/1809de155bad674b494248bcfca0e49eb4c5d8bee58f26fe7a0dd45029e2/numpy-1.15.4.zip (4.5MB)
Requirement already satisfied: flask<0.11,>=0.10.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.10.1)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (2.8.1)
Requirement already satisfied: regex==2018.11.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/regex-2018.11.3-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.7) (2018.11.3)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.8.3)
Requirement already satisfied: tornado==4.5.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/tornado-4.5.3-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.7) (4.5.3)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.7) (3.10)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.2.11)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.3.5)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.9)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.10)
Requirement already satisfied: markdown<2.7,>=2.6.9 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Markdown-2.6.11-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (2.6.11)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/pyserial-3.4-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (3.4)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.7.19)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.8.3)
Requirement already satisfied: sarge==0.1.4 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.1.4)
Requirement already satisfied: netifaces<0.11,>=0.10.6 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/netifaces-0.10.7-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.10.7)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (1.0.9)
Requirement already satisfied: rsa<3.3,>=3.2.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (3.2.3)
Requirement already satisfied: pkginfo<1.5,>=1.4.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/pkginfo-1.4.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (1.4.2)
Requirement already satisfied: requests<3,>=2.18.4 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/requests-2.20.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (2.20.1)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (2.4.2)
Requirement already satisfied: psutil<5.5,>=5.4.6 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/psutil-5.4.8-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.7) (5.4.8)
Requirement already satisfied: Click<6.3,>=6.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (6.2)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (1.6.5)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (5.2.1)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (1.0.2)
Requirement already satisfied: future<0.16,>=0.15.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.15.2)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.7) (1.3)
Requirement already satisfied: websocket-client<0.41,>=0.40.0 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.40.0)
Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/python_dateutil-2.6.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (2.6.1)
Requirement already satisfied: wrapt<1.11,>=1.10.11 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/wrapt-1.10.11-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.7) (1.10.11)
Requirement already satisfied: futures<3.2,>=3.1.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/futures-3.1.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (3.1.1)
Requirement already satisfied: emoji<0.5,>=0.4.5 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/emoji-0.4.5-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (0.4.5)
Requirement already satisfied: monotonic<1.4,>=1.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/monotonic-1.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (1.3)
Requirement already satisfied: frozendict<1.3,>=1.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/frozendict-1.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.7) (1.2)
Requirement already satisfied: itsdangerous>=0.21 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/itsdangerous-1.1.0-py2.7.egg (from flask<0.11,>=0.10.1->OctoPrint->Bed-Visualizer==0.1.7) (1.1.0)
Requirement already satisfied: MarkupSafe in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/MarkupSafe-1.1.0-py2.7-linux-armv7l.egg (from Jinja2<2.9,>=2.8.1->OctoPrint->Bed-Visualizer==0.1.7) (1.1.0)
[... cut ...]
  Using cached https://files.pythonhosted.org/packages/ff/47/1dfa4795e24fd6f93d5d58602dd716c3f101cfd5a77cd9acbe519b44a0a9/wheel-0.32.3-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.32.3
root@octopi:~# pip install numpy
Collecting numpy
  Using cached https://files.pythonhosted.org/packages/2d/80/1809de155bad674b494248bcfca0e49eb4c5d8bee58f26fe7a0dd45029e2/numpy-1.15.4.zip
Building wheels for collected packages: numpy
  Running setup.py bdist_wheel for numpy ... -

kantlivelong, 2018-12-02:
This tracker is reserved for:

- missing or broken homepage, source or download URLs
broken images
- incorrect compatibility information
- ... in short, anything related to the data on plugins.octoprint.org

If this issue only applies to a specific plugin please report to the appropriate plugin repository.

If this issue affects all plugins please submit a ticket to https://github.com/foosel/OctoPrint

closed 2018-12-02 by kantlivelong

techpad-marcb, 2018-12-06:
Oh sorry i clicked on the link provided in the plugin for reporting issues.