OctoPrint/plugins.octoprint.org #258: [TODO]  - Fix deprecated gem used in Travis CI

kantlivelong, 2018-12-12:
Noticed the following warning when reviewing a recent build failure:

`
$ bundle exec jsonlint ./_site/plugins.json
[DEPRECATION] This gem has been renamed to optimist and will no longer be supported. Please switch to optimist as soon as possible.
The command "bundle exec jsonlint ./_site/search.json" exited with 0.
`

foosel, 2018-12-19:
Not exactly sure if there's anything *we* can fix here... I just checked and `optimist` is a [command line option parser](https://rubygems.org/gems/optimist). My guess is either `bundler` or `jsonlint` depend on whatever this was called before.

kantlivelong, 2020-05-08:
This may no longer be relevant with the recent build changes. @foosel

foosel, 2020-05-08:
We are still using travis for the link testing and such, need to switch that over to actions too though. Same workflow should do.

foosel, 2020-05-24:
Now we no longer use travis.

closed 2020-05-24 by foosel