OctoPrint/plugins.octoprint.org #279: [Feature Request] Download count by plugin

servel333, 2019-02-18:
I would love to see some basic statistics on each plugin. Seeing how popular a plugin is helps me decide between several otherwise seemingly identical plugins.

For example, there are now 4 filament sensor plugins that all claim to pause the print. I have no idea how to tell what the difference between them is, nor make an informed decision without digging into the code of each one, and/or installing all of them and seeing how they behave.

Additionally, perhaps a feedback system would help too. Even a simple YouTube-style 👍 (Thumbs up) and 👎 (Thumbs down) and counts would be enough.

foosel, 2019-02-19:
While I agree that this would be some nice to have functionality, just a heads-up that it's not going to happen soon since it basically requires a full rewrite of the repository.

It's currently a simple static website that's living on Github Pages and being generated via Jekyll with some creative templating. What you are requesting here would require a shift to a dynamic page with a database for storing votes and such. And by that add an additional administrative overhead to maintain the servers. Which is what I so far actually wanted to avoid.

Maybe there's a possibility to utilize the community forum as a backend, I've been thinking about that for a while, but so far haven't looked closer into it. Same goes for some nginx magic and automated log parsing to achieve a basic download counter.

servel333, 2019-02-19:
Having explained that, I can see how big of a problem this is.

I was hoping you could at least pull download counts from Github, but I don't see counts on the release download page, so I'm not sure if that data is publicly available.

I can't think of any way to do a rating system without a server. You could have JS that makes an API request to a server to get counts and apply ratings, so the site could still be hosted on Github pages, but you would still need an API somewhere. There are free options I can think of, and some are easier, but nothing that is totally maintenance free that I know of.

Thanks for the feedback. Perhaps this isn't feasible.

foosel, 2019-02-20:
The problem with a download counter through Github releases (which last time I checked wasn't possible via API) is that it also only would work with plugins released on Github. Not all of them are, nor should they have to be.

Easiest approach (with added administrative overhead) would be a download endpoint on plugins.octoprint.org, redirecting to the installable package but counting how often which one is being requested. The backend code needed for that *should* be fairly straight forward.

I've even thought about a custom python package index for the registered plugins, to also allow wheels of the plugins and their dependencies to be prebuilt and all that (of course, that means maintenance overhead). But that's so far also just been on the level of shower thoughts. 

Voting is a tad more involved. It shouldn't be anonymous and/or cookie based, since you want to avoid one person being able to upvote something multiple times. Which means now you are looking at account management. Which also means GDPR requirements just kicked in on a tighter level ;)

In a nutshell, I'm not saying that this will never happen or that it's completely unfeasible. It's just sitting at a very very low priority since it's not a low hanging fruit and there are a ton of other things on my plate. I wouldn't say no to a helping hand though ;)

benlye, 2019-04-24:
Related request - could you publish plugin installation statistics based on the anonymous usage tracking data that OctoPrint sends back?  As a plugin developer it would be nice to know how many users have my plugins installed. Version data would be nice too. Not looking for anything to detailed, just something like 'Installed on X Systems'.

foosel, 2019-04-24:
@benlye I'm currently not able to track already installed plugins, only plugin installs. Reason is the way that tracking currently works (GET requests, limited URL length, hence limited payload data not fitting a full blown plugin list). I'm actively looking into this but I'm not 100% happy with the implementation so far.

OllisGit, 2019-09-30:
Hi @servel333 , @foosel ,

I had the same issue and my solution for me as a developer is to add a download counter on my github-pages. E.g. https://github.com/OllisGit/OctoPrint-DisplayLayerProgress

It is a little bit tricky, because github didn't count downloads from source-zip files (as foosel already mentioned). It only counts downloads when they have have additional "assets" in a  release.
I described my solution in the octoprint-forum: https://community.octoprint.org/t/plugin-download-statistics/9366

Short: 
- After each commit a Draft-Release is build with an attached asset (done by travis)
- In the plugin you must change the download location from "source.zip" to "release-asset.zip"

A detailed description is also available in my Knowledge-Wiki: https://github.com/OllisGit/OctoPrint-KnowledgeBase/wiki/DownloadCounter

Works pretty good! 

So, maybe you @foosel can do a little bit of "marketing" for this solution..if you like ;-) Then the user could look into the github-pages and see the popularity of a plugin.

What do you think?
Olli

foosel, 2020-05-13:
Thanks to Github Actions it's now possible to do a bit of page build magic, and over the course of this day I put together a solution that retrieves the installation stats from the tracking plugin and uses them to add installation counts (which are more meaningful IMHO than download counts anyhow) to the plugin detail pages, and also a Top 10. 

I'm closing this as this should hopefully provide the wanted for indication of utilization of the various plugins, without a huge amount of additional overhead that download counters would add.

closed 2020-05-13 by foosel

benlye, 2020-05-14:
That's perfect - thanks 👍