OctoPrint/plugins.octoprint.org #280: ShutdownPrinter

wimorrison, 2019-02-22:
when you try to install this plugin 

https://plugins.octoprint.org/plugins/shutdownprinter/

You get this error

Cannot unpack file /tmp/pip-unpack-__pJK0/shutdownprinter (downloaded from /tmp/pip-req-build-3MRHuC, content-type: text/html; charset=utf-8); cannot detect archive format
Cannot determine archive format of /tmp/pip-req-build-3MRHuC

Great pity as it looks like a good plugin - can it be fixed please?

foosel, 2019-02-25:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin **repository**, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
>
>   * missing or broken homepage, source or download URLs
>   * broken images
>   * incorrect compatibility information
>   * ... in short, anything related to the data on plugins.octoprint.org
>
> What you should not report here:
>
>   * Any kinds of issues with the plugins themselves - report those
>     to the plugin author instead!

Thanks :)

closed 2019-02-25 by foosel

wimorrison, 2019-02-25:
Can you print me at the thread to log the issue in - I looked and want able to find one hence posting it here

foosel, 2019-02-25:
The plugin's homepage and repository is at https://github.com/devildant/OctoPrint-ShutdownPrinter

It's prominently linked to from the very same URL you posted in your initial post ;)