OctoPrint/plugins.octoprint.org #311: LEDStripControl - ERROR - Unable to communicate with PiGPIOd

Supersizeme2007, 2019-05-14:
After update octoprint to 1.3.11

LEDStripControl - ERROR - Unable to communicate with PiGPIOd in LOG-File.

kantlivelong, 2019-05-14:
This tracker is reserved for:

- missing or broken homepage, source or download URLs
broken images
- incorrect compatibility information
- ... in short, anything related to the data on plugins.octoprint.org

If this issue only applies to a specific plugin please report to the appropriate plugin repository.

If this issue affects all plugins please submit a ticket to https://github.com/foosel/OctoPrint

closed 2019-05-14 by kantlivelong

Supersizeme2007, 2019-05-15:
Hi
It is only the Plugin LEDStripControl. 
If i deaktivate it all normal!

Von: Shawn Bruce
Gesendet: Dienstag, 14. Mai 2019 23:52
An: OctoPrint/plugins.octoprint.org
Cc: Supersizeme2007; Author
Betreff: Re: [OctoPrint/plugins.octoprint.org] LEDStripControl - ERROR -Unable to communicate with PiGPIOd (#311)

This tracker is reserved for:
• missing or broken homepage, source or download URLs
broken images
• incorrect compatibility information
• ... in short, anything related to the data on plugins.octoprint.org
If this issue only applies to a specific plugin please report to the appropriate plugin repository.
If this issue affects all plugins please submit a ticket to https://github.com/foosel/OctoPrint