OctoPrint/plugins.octoprint.org #318: [TODO] [Feature Request] Raspberry Pi 3B+ to turn off USB power on disconnect

ozgunawesome, 2019-05-26:
Hi folks,

My OctoPrint is running on a Raspi 3B+ and connects to my Ender3 via USB serial.

Ender3 has a manual switch power supply but I have a smart relay to turn it on and off.

When I turn off the power supply, the Raspi still keeps powering the printer via the 5V rail. This results in:

* LED backlight always on (but the display font is very faint)
* BLtouch always powered on
* If I send a print through Ultimaker Cura it actually tries to print it (but can't get temps up because the 24V rail is not powered)

So I wrote a plugin to power down Raspi USB when printer disconnects, and powering it back up when you click Connect on OctoPrint UI

Source available here: https://github.com/ozgunawesome/OctoPrint-USBPowerControl

Disconnect works fine, on the "disconnect" event the port powers down, but for "connecting" event there are following issues:

* Serial port needs to be set to AUTO
* You gotta click "Connect" twice. Because on first click it powers the USB up and the kernel needs some time to enumerate the /dev/ttyUSB devices that newly appear. Meanwhile you get a serial port enumeration error. Second click on "connect" it works fine.

I think I need to hook another event instead of printer connection status.

@foosel since you seem to be the lead on this repo with the most commits can you give me some guidance here? I basically need a 1-2 second delay after clicking "connect" so the Raspi USB hardware has time to power up and enumerate the serial devices.

I'd like to fix this issue then merge it upstream.

Thanks for the help in advance..

Özgün

foosel, 2019-05-27:
This would actually better asked on the [community forum](https://community.octoprint.org), more specifically the "development" category. The bug tracker is reserved for issues with the plugin repository.

To quickly give you a hint here though - the problem is that events run asynchronously, so the power up might even happen after the connection attempt depending on how much stuff is backlogged in the event bus. What you want instead is actually wrap the connection method itself. There's no official hook for that or something, but it can still be done with a big of monkey patching. Take a look at [this plugin](https://github.com/foosel/OctoPrint-PbPiLink/blob/198e5d78b6643a47f4aa7b932be87da0e13787ca/octoprint_pbpilink/__init__.py#L39-L45) that does it.

>     * Serial port needs to be set to AUTO

I take it this is because of the serial port only showing up after firing up the power again? Tricky to solve then - no way to show a device in a selection box which ain't there yet, and `AUTO` can be unreliable if there's more than one serial interface detected.

ozgunawesome, 2019-05-28:
> This would actually better asked on the [community forum](https://community.octoprint.org), more specifically the "development" category. The bug tracker is reserved for issues with the plugin [...]

Thanks for the sample code, that's exactly what I needed. You're right on the serial port `AUTO` because there's a small delay between powering up Raspberry Pi USB ports and the /dev/tty devices appearing.

In my current implementation, the first click on "Connect" button powers up the hub but since it's asynchronous, the connection fails before the port appears. The second click then works fine.

I'll instead override the Connect method itself. Thanks for the pointer :)

closed 2019-05-28 by ozgunawesome