OctoPrint/plugins.octoprint.org #324: Fails to install

Neiljclements, 2019-05-31:
Firstly, it doesn't appear in a search within the plug in manager, so I attempted to install via URL within plug in manager.  However, I get the error as shown in the photo.

[image]

foosel, 2019-05-31:
a) That's the wrong URL, see https://plugins.octoprint.org/plugins/octopod/ for the right one: `https://github.com/gdombiak/OctoPrint-OctoPod/archive/master.zip` 

b) You might have to refresh the plugin list from the repository for it to show up there:

[image]

foosel, 2019-05-31:
Just tested on a current 1.3.11 instance:

[image]

Needs a refresh though, see above. OctoPrint caches the repository data for 24h, unless you have something else configured:

[image]

closed 2019-05-31 by foosel

Neiljclements, 2019-05-31:
Thank you very much.  It works perfectly now.  I'm a big fan of OctoPod and I'm sure this will compliment it perfectly.

gdombiak, 2019-06-02:
FYI @foosel, I'm getting reports that it has been over 48 hours and the plugin does not appear in the local OctoPrint instance. Users have to click on refresh list from repo as mentioned above to make the plugin appear. I have no logs or other info.

foosel, 2019-06-03:
Hm... I can reproduce this on an instance I haven't touched in several weeks and am very much confused by this. I did not change anything on the server or on the build. 

*sigh* this is going to be fun to figure out

foosel, 2019-06-03:
Seems to be a bug in OctoPrint's Plugin Manager that's been in there forever. It should refresh the cache on repository requests from the client/browser, but this appears to never have been implemented in that way. See https://github.com/foosel/OctoPrint/issues/3190. Funny that it only now got noticed.

gdombiak, 2019-06-03:
Thanks @foosel, great detective work!

Neiljclements, 2019-06-04:
This app is awesome!
Where would be the place to leave suggestions for additions to the app? I’d love to have information like what layer was being printed on the first panel screen and a couple of other ideas.

gdombiak, 2019-06-04:
Thanks @Neiljclements for the nice words!. Here is the main [git repo for OctoPod](https://github.com/gdombiak/OctoPod) and you can file suggestions directly [here](https://github.com/gdombiak/OctoPod/issues).

Regarding current Z layer, I wanted that feature too when I started OctoPod. I even coded it to later realize that OctoPrint removed support for it since it was not reporting an accurate number. I think that the issue is the Marlin firmware not reporting it so OctoPrint cannot be certain. IIRC, there is a plugin that tried to restore that feature but it was not accurate due to the same reason. If the information is available in OctoPrint then I'd add it in a heartbeat. :)

Thanks,
Gaston