OctoPrint/plugins.octoprint.org #337: Plugin TouchUI should be on blacklist

mkh595, 2019-08-15:
TouchUI doesn't work from OctoPrint version 1.3.10 and caused that OctoPrint cannot be open neither in web on PC. Issues are for several months without any progress.

BillyBlaze, 2019-08-16:
There are alot of users who don't have the same issue as you and are happily using TouchUI. Next time if you want somebody to look into your issue, don't say stuff like "Many thanks for such piece of junk.", you can uninstall TouchUI if you don't like it, nobody is forcing you to install it.

jubaleth, 2019-08-16:
Hi,

This tracker is reserved for issues related to the plugin repository *only*. If you're looking for support, community.octoprint.org has a large user base who may be inclined to help you if you respectfully ask for help. As @BillyBlaze said, there are a lot of users who are using the UI without the issue you're having, this is *not* a reason to blacklist a plugin. 

Moreover, I suggest that when you do ask for support (on the right medium) that you will provide more information than "it doesn't work". With the current level of information provided, nobody can help you with your problem.

Good luck.

closed 2019-08-16 by jubaleth