OctoPrint/plugins.octoprint.org #351: Plugin Version 2.0.0 not updating/installing on Version Octopi 1.3.12rc3

ZeekWorld, 2019-09-24:
I've had installed 1.0.0 and was working without issues. An update became available 2.0.0, update failed to installed.
I uninstalled the plug in and re installed. Still no luck. See error below.
Running Octoprint (OctoPi) Version 1.3.12rc3

Installing plugin "Printer Statistics" from https://github.com/amsbr/OctoPrint-Stats/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/amsbr/OctoPrint-Stats/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
Downloading https://github.com/amsbr/OctoPrint-Stats/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Stats==2.0.0) (1.3.12rc3)
Collecting tinydb (from OctoPrint-Stats==2.0.0)
Downloading https://files.pythonhosted.org/packages/dc/90/ced9f2a57513eb75dce70760e55967ceedbf2dc158b6a66f1bf87ac8fab0/tinydb-3.14.1-py2.py3-none-any.whl
Collecting pandas (from OctoPrint-Stats==2.0.0)
Downloading https://files.pythonhosted.org/packages/b2/4c/b6f966ac91c5670ba4ef0b0b5613b5379e3c7abdfad4e7b89a87d73bae13/pandas-0.24.2.tar.gz (11.8MB)
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-umQrj2/pandas/setup.py", line 746, in <module>
**setuptools_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 144, in setup
_install_setup_requires(attrs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
dist.fetch_build_eggs(dist.setup_requires)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 724, in fetch_build_eggs
replace_conflicting=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 846, in resolve
dist = best[req.key] = env.best_match(req, ws, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1118, in best_match
return self.obtain(req, installer)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1130, in obtain
return installer(requirement)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 791, in fetch_build_egg
return cmd.easy_install(req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
return self.install_item(spec, dist.location, tmpdir, deps)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
dists = self.install_eggs(spec, download, tmpdir)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
return self.build_and_install(setup_script, setup_base)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
self.run_setup(setup_script, setup_base, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
run_setup(setup_script, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
raise
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
saved_exc.resume()
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
six.reraise(type, exc, self._tb)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
yield saved
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
yield
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
_execfile(setup_script, ns)
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
exec(code, globals, locals)
File "/tmp/easy_install-nQl7wJ/numpy-1.17.2/setup.py", line 31, in <module>
RuntimeError: Python version >= 3.5 required.
**----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-umQrj2/pandas/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output**

foosel, 2019-09-24:
That's an issue with the plugin, it states that it requires a python version that btw isn't even supported by stable OctoPrint yet. You'll need to report that in the plugin's tracker, I can't solve that here.

closed 2019-09-24 by foosel

ZeekWorld, 2019-09-24:
ok thanks.