OctoPrint/plugins.octoprint.org #381: [Feature Request] Help me avoid stale plugins

stantond, 2019-12-12:
As a user, I want to install maintained plugins, so that I do not cause problems with my print process by introducing outdated and unmaintained code.

As a user, I want to install the latest stable version of a plugin, even if it comes from a different contributor's fork, so that I do not cause problems with my print process by introducing outdated and unmaintained code.

Suggestions:
- Regularly update the OctoPrint plugin pages with data from GitHub (where available), including `Last Commit` and `Number of Open Issues`.
- Create a mechanism for a contributor to take over the official listing for a plugin and point it to their fork when the original owner is not interested in continuing development. A recent example of this is [OctoPrint-DetailedProgress](https://github.com/dattas/OctoPrint-DetailedProgress). There are 24 open issues and 8 PRs. The original creator no longer maintains the project, and [hasn't responded to a request to allow someone else to maintain it](https://github.com/dattas/OctoPrint-DetailedProgress/issues/40). If you review the PRs and various forks, there are some great features and fixes spread around, but they are not consolidated into a single repo. It feels fair to assume that most OctoPrint users will trust plugins.octoprint.org, and will never realise that they are installing out of date plugins with open issues (bugs).

foosel, 2019-12-12:
Just wanted to quickly say: I agree that that's a problem that should be looked into, I just don't have the time at the moment for that until 1.4.0 stable is out (hopefully very early 2020).

tedder, 2020-03-30:
Linking the discussion: https://community.octoprint.org/t/plugin-compatibility-in-plugin-manager/17636

foosel, 2020-05-14:
>     * Regularly update the OctoPrint plugin pages with data from GitHub (where available), including `Last Commit` and `Number of Open Issues`.

This is now done.

jneilliii, 2020-09-21:
I think the second suggestion posted here is being fixed in the next release of OctoPrint 1.5.0 is it not @foosel?

foosel, 2020-09-21:
I think you are right, thanks for the reminder! 

Mechanism in place will be the ability to override the update configuration in future OctoPrint versions based on data stored in the plugin repository (though `config.yaml` will always win) via an [overlay](https://plugins.octoprint.org/overlays/). Already working with OctoPrint's `maintenance` branch/future 1.5.0.

closed 2020-09-21 by foosel