OctoPrint/plugins.octoprint.org #388: Having issues distributing plugin

NerdboyQ, 2020-01-08:
Hello, 

I have successfully developed, tested, and created a for for my app. However I am trying to go through the steps to merge my fork into the master but with no success. 
My fork : https://github.com/NerdboyQ/plugins.octoprint.org

I have fetched the changes, ran my merge (origin -> remote) and pushed. However, I do not see my md file in this repository. Is there a waiting period for the status of my submission?

[I'm not sure if this is the correct place to reach out but I didn't see anywhere else]

foosel, 2020-01-31:
Apologies for the late reply, this slipped through.

It doesn't suffice to push to your fork, you also need to create a so called pull request to this repository here, which Github should support you with.

See here: https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork

closed 2020-01-31 by foosel

NerdboyQ, 2020-01-31:
Thanks a lot!