OctoPrint/plugins.octoprint.org #393: problem with raspberry pi zero W and Prusa i3 mk3s

gasrovescio, 2020-01-29:
READ FIRST, THEN DELETE:

What kinds of issues you should report here:

  * missing or broken homepage, source or download URLs
  * broken images
  * incorrect compatibility information
  * ... in short, anything related to the data on plugins.octoprint.org

What you should not report here:

  * Any kinds of issues with the plugins themselves - report those
    to the plugin author instead!

Thanks :)

gasrovescio, 2020-01-29:
hi, I am writing to you because I have a problem installing the plug-in
  Bed Level Visualizer, on my raspberry PI zero W. The installation does not take place due to an unknown problem.

Installing plugin "Bed Level Visualizer" from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip --no-cache-dir
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
Downloading https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Bed-Visualizer==0.1.11)
Collecting numpy (from Bed-Visualizer==0.1.11)
Downloading https://files.pythonhosted.org/packages/b7/6f/24647f014eef9b67a24adfcbcd4f4928349b4a0f8393b3d7fe648d4d2de3/numpy-1.16.6.zip (5.1MB)
Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: Flask-Principal<0.5,>=0.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: werkzeug<0.16,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: markdown<3.1,>=3.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: netifaces<0.11,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: pylru<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: rsa<5,>=4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: requests<3,>=2.21.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: semantic_version<2.7,>=2.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: psutil<5.7,>=5.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: Click<8,>=7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: chainmap<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: future<0.18,>=0.17.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: scandir<1.11,>=1.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: websocket-client<0.57,>=0.56 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: wrapt<1.12,>=1.11.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
Requirement already satisfied: futures<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.11)
[... cut ...]
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Error!
Could not install plugin, unknown error, please consult octoprint.log for details

foosel, 2020-01-30:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin **repository**, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
>
>   * missing or broken homepage, source or download URLs
>   * broken images
>   * incorrect compatibility information
>   * ... in short, anything related to [...]

Thanks :)

Additional be advised that contrary to what Prusa Research advises, the Raspberry Pi 0w is **not recommended** for use with OctoPrint as it is severely underpowered.

closed 2020-01-30 by foosel