OctoPrint/plugins.octoprint.org #411: [Abandoned Plugin] M33Fio

foosel, 2020-03-18:
It looks like the [M33-Fio plugin](https://github.com/donovan6000/M33-Fio) has been abandoned by its author and is no longer maintained. Due to that it has been disabled on the repository for now.

See https://github.com/donovan6000/M33-Fio/issues/256#issuecomment-599010588

If you rely on this plugin and would be interested in taking over maintenance, please step up.

welshjohn60, 2020-03-18:
Hi @foosel. This has just popped up a box in Octoprint saying "Notice regarding plugin "Anonymous Usage Tracking"" - which I am guessing you didn't want to happen!

[image]

foosel, 2020-03-18:
True... ARgh. Thanks for the heads-up. Should be fixed.

foosel, 2020-03-18:
Of course it will already have confused some people at this point. And might still confused more due to caching. I've pushed a notice into the OP, but that's about all I can do for now. Damn... 

I tested the notification mechanism before creating this ticket, and for that quickly used a bundled plugin, and then of course forgot to change out the identifier prior to committing. Bloody hell.

DataWorm, 2020-04-06:
This plugin actually bundled quite a lot of more or less different features. Maybe it would help to split it up into several plugins so that at least some of those features might continue working or that volunteers only need to maintain a smaller part of this current plugin?!

turican0, 2020-04-06:
This plug-in is only one for use multiple printers on single octoprint, with settings via interface.

mod38, 2020-04-16:
Hi
I use OctoPi 0.15.1 with Octoprint 1.3.10 to drive an original M3D Micro. I have not upgraded to any later version as I used the embedded Cura version which works fine for me. If anything happens the SD card on the Pi I am using, I will have to reimage the card or a new one with that version of OctoPi but I won't be able to use it if I can't download the M33 Fio plugin. Is there a real problem with leaving it available? While Donovan6000 will not be updating it the combination I am using does not seem to require any maintenance and works brilliantly. My only other option would be to use M3D's own software which I use on one of their Micro+ machines but the functionality of the Octoprint/M33Fio combination is much better giving more choices in setting up the print as well as using a camera on the Pi - which is one of the main reasons for the existence of OctoPrint!
I hope you can reenable the plug-in, if necessary with any warning you see as necessary.

foosel, 2020-04-16:
The plugin is still available and can be manually installed via its URL through plugin manager:

[image]

It's merely disabled on the repository, disabling it there doesn't make it magically vanish from the internet.

> but the functionality of the Octoprint/M33Fio combination is much better giving more choices in setting up the print as well as using a camera on the Pi

I really wish someone would step up to adopt it then... sadly so far that hasn't happened.

foosel, 2020-04-16:
> I have not upgraded to any later version as I used the embedded Cura version which works fine for me.

For the record, that ALSO still works, you just have to install the CuraLegacy plugin on later OctoPrint versions...

mod38, 2020-04-16:
Hi Gina
Thanks for this quick response. I am not completely clear yet however. Are you saying that I won't be able to install it through the plugin manager but could use the pip install you show above ( in terminal on the Pi)?
I tried the CuraLegacy plugin when it was issued but it didn't work very well for me so I reverted to earlier versions. The original M3D Micro won't last forever so I'll keep going as I am.
I am waiting to move to a Prusa Mini when it settles down and of course that beings up another set of Octoprint questions....

foosel, 2020-04-16:
I'm saying that while you cannot install it from the list under (1) you can still install it by copy pasting the URL into (2):

[image]

> I tried the CuraLegacy plugin when it was issued but it didn't work very well for me so I reverted to earlier versions.

It is literally the same code as in earlier versions, just extracted. It would have been nice to know what kind of issues you experienced with it in form of a bug report on its repository.

mod38, 2020-04-16:
It was last year and I think I figured it was just that I didn't have time to sort it out. As far as I remember my problem was that the installation asked me to enter a printer profile and I couldn't see how to do that. While I use Raspberry Pis and do some 3D printing I am not a Linux expert or a printing expert so I left it aside. I think the updates to Octoprint from 1.3.10 didn't seem to include anything I needed so I didn't feel I was missing out.
Now that I can't go out much of course I may give it another try....

foosel, 2021-07-30:
Plugin has been delisted, see #470

commit fc9349f3 (2021-07-30): Delist m33fio

Closes #411
Closes #470
closed 2021-07-30 by foosel