OctoPrint/plugins.octoprint.org #413: Installation Problem

Neiljclements, 2020-03-25:
I have an issue installing.  It sits at the screen shown in the picture and does no more. I've tried two of my OctoPrint servers and they both did the same.

[image]

foosel, 2020-03-25:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin **repository**, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
>
>   * missing or broken homepage, source or download URLs
>   * broken images
>   * incorrect compatibility information
>   * ... in short, anything related to the data on plugins.octoprint.org
>
> What you should not report here:
>
>   * Any kinds of issues with the plugins themselves - report those
>     to the plugin author instead!

Thanks :)

closed 2020-03-25 by foosel

Neiljclements, 2020-03-25:
Apologies