OctoPrint/plugins.octoprint.org #414: Can't install Plugin octoprint-LCD1602

VASAPOL, 2020-03-26:
ERROR CODE:
Installing plugin "octoprint-LCD1602" from https://github.com/n3bojs4/OctoPrint-Lcd1602/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/n3bojs4/OctoPrint-Lcd1602/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/n3bojs4/OctoPrint-Lcd1602/archive/master.zip
Downloading https://github.com/n3bojs4/OctoPrint-Lcd1602/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Lcd1602==0.1.0) (1.4.0)
Collecting RPLCD
Downloading https://www.piwheels.org/simple/rplcd/RPLCD-1.2.2-py2.py3-none-any.whl (45kB)
Collecting smbus2
Downloading https://www.piwheels.org/simple/smbus2/smbus2-0.3.0-py2.py3-none-any.whl
Collecting fake-rpi
Downloading https://files.pythonhosted.org/packages/2f/e9/23664caf50766e4a082c8981d2333e7b03237205bb85782e3cb3d70519d5/fake_rpi-0.6.2-py2-none-any.whl
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.6.5)
Requirement already satisfied: blinker<1.5,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.4)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.12.2)
Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.12)
Requirement already satisfied: netifaces<0.11,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.10.9)
Requirement already satisfied: sentry-sdk==0.13.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.13.2)
Requirement already satisfied: requests<3,>=2.22.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (2.22.0)
Requirement already satisfied: Click<8,>=7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (7.0)
Requirement already satisfied: pylru<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.2.0)
Requirement already satisfied: Flask-Login<0.5,>=0.4.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.4.1)
Requirement already satisfied: wrapt<1.12,>=1.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.11.2)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (3.1.1)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.1)
Requirement already satisfied: websocket-client<0.57,>=0.56 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.56.0)
Requirement already satisfied: chainmap<1.1,>=1.0.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.0.3)
Requirement already satisfied: future<0.19,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.18.2)
Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (4.5.3)
Requirement already satisfied: semantic-version<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (2.8.4)
Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.5.0.1)
Requirement already satisfied: futures<3.4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (3.3.0)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.7.19)
Requirement already satisfied: werkzeug<0.17,>=0.16 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.16.1)
Requirement already satisfied: flask<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.12.5)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (2.8.1)
Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.2)
Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.9.0)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (2019.8.19)
Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (3.7.4.1)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.1.5.post0)
Requirement already satisfied: psutil<5.7,>=5.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (5.6.7)
Requirement already satisfied: filetype<2,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.0.5)
Requirement already satisfied: emoji<0.6,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.5.4)
Requirement already satisfied: monotonic<1.6,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.5)
Requirement already satisfied: rsa<5,>=4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (4.0)
Requirement already satisfied: scandir<1.11,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.10.0)
Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (5.1.2)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (5.2.1)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Lcd1602==0.1.0) (3.4)
Collecting numpy
Downloading https://files.pythonhosted.org/packages/b7/6f/24647f014eef9b67a24adfcbcd4f4928349b4a0f8393b3d7fe648d4d2de3/numpy-1.16.6.zip (5.1MB)
Collecting build-utils
Downloading https://files.pythonhosted.org/packages/42/79/014c9458b32861201a25157c478d2c73f00c493861b822d3b914fa68a799/build_utils-0.3.2-py2-none-any.whl
Requirement already satisfied: Unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from awesome-slugify<1.7,>=1.6.5->OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.4.21)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<0.13,>=0.12->OctoPrint->OctoPrint-Lcd1602==0.1.0) (2.7.0)
Requirement already satisfied: webassets>=0.11.1 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Assets<0.13,>=0.12->OctoPrint->OctoPrint-Lcd1602==0.1.0) (0.12.1)
Requirement already satisfied: urllib3>=1.10.0 in /home/pi/oprint/lib/python2.7/site-packages (from sentry-sdk==0.13.2->OctoPrint->OctoPrint-Lcd1602==0.1.0) (1.25.6)
Requirement already satisfied: certifi in /home/pi/oprint/lib/python2.7/site-packages (from sentry-sdk==0.13.2->OctoPrint->OctoPrint-Lcd1602==0.1.0) (2019.9.11)
Requirement already satisfied: idna<2.9,>=2.5 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.22.0->OctoPrint->OctoPrint-Lcd1602==0.1.0) (2.8)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.22.0->OctoPrint->OctoPrint-Lcd1602==0.1.0) (3.0.4)
Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python2.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->OctoPrint-Lcd1602==0.1.0) (41.5.1)
[... cut ...]
Downloading https://files.pythonhosted.org/packages/81/44/8a15e45ffa96e6cf82956dd8d7af9e666357e16b0d93b253903475ee947f/docutils-0.16-py2.py3-none-any.whl (548kB)
Collecting Pygments>=2.5.1
Downloading https://www.piwheels.org/simple/pygments/Pygments-2.6.1-py2.py3-none-any.whl (914kB)
ERROR: Package 'Pygments' requires a different Python: 2.7.16 not in '>=3.5'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Raspberry pi 4 model 4 GB

Please help me to fix these issue.

foosel, 2020-03-26:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin **repository**, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
>
>   * missing or broken homepage, source or download URLs
>   * broken images
>   * incorrect compatibility information
>   * ... in short, anything related to the data on plugins.octoprint.org
>
> What you should not report here:
>
>   * Any kinds of issues with the plugins themselves - report those
>     to the plugin author instead!

Thanks :)

closed 2020-03-26 by foosel

VASAPOL, 2020-03-26:
I can't install plugin plugin link https://plugins.octoprint.org/plugins/LCD1602_I2cdisplay/
ERROR Code is

ERROR: Package 'Pygments' requires a different Python: 2.7.16 not in '>=3.5'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

foosel, 2020-03-26:
That's an issue the *author of the plugin* has to fix, who's not me. Get in touch with the plugin author!