OctoPrint/plugins.octoprint.org #418: Can't send data to plot.ly anymore

IroCorp, 2020-03-29:
I can't update data to plot.ly anymore. I just have the option to save to PNG.
Maybe because of the change of the URL? 
ttps://community.plotly.com/t/url-change-for-chart-studio-cloud-https-chart-studio-plot-ly/33557

Thank you

foosel, 2020-03-29:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin **repository**, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
>
>   * missing or broken homepage, source or download URLs
>   * broken images
>   * incorrect compatibility information
>   * ... in short, anything related to the data on plugins.octoprint.org
>
> What you should not report here:
>
>   * Any kinds of issues with the plugins themselves - report those
>     to the plugin author instead!

Thanks :)

closed 2020-03-29 by foosel

IroCorp, 2020-03-29:
OK Sorry :)