OctoPrint/plugins.octoprint.org #434: [Compatibility] OctoPrint-Stats requires Python >= 3.5

VonOgre, 2020-04-15:
The latest release of OctoPrint-Stats (2.0.2) fails to install because it requires Python >= 3.5, though the page listing says >=2.7, < 3.

It seems to have to do with the addition of Pandas and subsequently Numpy, the latter of which requires Python 3.5 or newer.
[plugin_pluginmanager_console.log](https://github.com/OctoPrint/plugins.octoprint.org/files/4481567/plugin_pluginmanager_console.log)

foosel, 2020-04-15:
The problem is that the plugin itself lacks a python compatibility flag, so setting it to python 3 here would make it completely uninstallable. The author needs to either pin their dependency to something that's still Python 2 compatible or update their plugin to python 3 and python 3 **only** readiness, including the compatibility flags in both the source and the listing here. Please report this on the plugin's own issue tracker (possibly in ambsr/OctoPrint-Stats#36 though that has been closed for some weird reason). For now I'm marking this plugin as disabled due to these problems.

If the plugin appears to be abandoned by the maintainer I can mark it as such on request. This should first be verified however by asking in the plugin's own tracker.

VonOgre, 2020-04-15:
I'll submit an issue there, since it seems like the issue you linked was resolved by only supporting python 3, but failed to take the compatibility flags into account.

foosel, 2020-04-15:
It actually never got the Python 3 flag in the plugin source. The ticket was closed after the one who opened it patched the source themselves. The author still hasn't reacted.

Davide2708, 2020-04-23:
Today is there any solution to install this plugin? I have octoprint version 1.4.0 but I can't install it. solutions?

VonOgre, 2020-04-23:
@Davide2708 - If you're currently running 1.4.0 on Python 2, your only solution would be to [do the migration](https://docs.octoprint.org/en/master/plugins/python3_migration.html) to run it on Python 3 and ensure that you have at least Python 3.5 installed. I haven't yet done it, as this plugin wasn't critical to me and was more of a convenience.

foosel, 2020-04-23:
Since the settings are not python specific, switching to a py3 venv would get you up and running quickly. Minus plugins - those would need to be reinstalled, if they are even yet available for python3 which is going to be the biggest challenge for a while yet.

Davide2708, 2020-04-23:
> @Davide2708 - If you're currently running 1.4.0 on Python 2, your only solution would be to ~[do the migration](https://docs.octoprint.org/en/master/plugins/python3_migration.html)~ migrate or reinstall OctoPrint running on Python 3 and ensure that you have at least Python 3.5 installed. I haven't yet done it, as this plugin wasn't critical to me and was more of a convenience.
> 
> @foosel - Is there an in-place migration from Python 2 to Python 3 for Octoprint 1.4.0 or does it involve a fresh installation for it to work?

how do you migrate or reinstall with python 3? and do you think this problem will be solved without changing the whole system? thank you very much

foosel, 2020-04-29:
There's a note on environment migration here: https://docs.octoprint.org/en/master/plugins/python3_migration.html.

Alternatively, manually installing an older version of numpy into the py2 environment (`pip install numpy==<version number>`) might also do the trick.

foosel, 2020-05-18:
Plugin is apparently abandoned, see #479. Closing this - it's marked as disabled and that's about all that can be done for now until someone adopts this.

closed 2020-05-18 by foosel