OctoPrint/plugins.octoprint.org #438: [Listing Error]

justinmbrock, 2020-04-18:
**Which Plugin? (URL to plugin page on plugin repository)**
https://plugins.octoprint.org/plugins/thespaghettidetective/

**What's wrong with the plugin *listing*?**
This plugin loads an iframe with a YouTube video in it that includes multiple trackers. There is no way to opt out of the tracking or disable loading of the YouTube video. The iframe also severely degrades performance on low end systems. The developer will not consider removing the video. It serves no technical purpose with regards to the functioning of the plugin.

jneilliii, 2020-04-18:
@justinmbrock, I just posted over on that plugin's repository with a possible solution to this issue.

https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/issues/51

foosel, 2020-04-20:
You've reported a listing error, however the issue appears to be with the plugin itself and not the listing (the youtube embeds on that one are click-to-embed). Issues with plugins need to be reported to the plugin author, like already done here.

I've commented on that plugin however.

justinmbrock, 2020-04-20:
I reported it here because the plugin loads a Youtube video that in turn loads imbedded trackers. There is no mechanism to opt out of that tracking, which contradicts Octoprint's instructions for registering a plugin on the official repository.

Mirarkitty, 2020-04-26:
I also would say this should not be in the official plugin repository until it's fixed. For them it's a "philosophical debate". For many of us it's "unacceptable".

jneilliii, 2020-04-26:
I don't know if it's released yet, but the plugin author was making the change to load the privacy url for the video.

foosel, 2020-04-27:
AFAIK the release is still planned to happen.

> I also would say this should not be in the official plugin repository until it's fixed. For them it's a "philosophical debate". For many of us it's "unacceptable".

In my personal experience it's better to address issues like this in a diplomatic way by seeking a dialogue instead of creating an incident and potentially damaging the ecosystem and poisoning the communication patterns for everyone. In this specific case I've reached out to the plugin author, explained to them why the current implementation as it is is not acceptable and needs to be changed and they agreed and promised to do the requested changes in the next scheduled release. 

Unlisting it now would send the wrong message. I've added a note to the listing however. Also reopening this ticket which has now been relabeled to reflect the actual issue on hand, will close it once the release with the fix is out. 

Also CC @kennethjiang for transparency.

kennethjiang, 2020-04-27:
Thank you @foosel for being unbiased. And @jneilliii for pitching in the PR. The PR will be released in just a few days with sentry opt-in.

justinmbrock, 2020-04-28:
Thank you @kennethjiang for accepting @jneilliii 's PR. @foosel, I understand where you are coming from. That PR written by @jneilliii wouldn't have been submitted to fix this if I hadn't posted here after @kennethjiang closed multiple reports for this from multiple people. After it was addressed and dismissed several times, I felt reporting it here was warranted because it was listed officially.

tedder, 2020-04-29:
That's not terribly kind. It took some intervention but Kenneth is good folk, as is Gina and jnielliii. Understandings are common and that's okay.

kennethjiang, 2020-04-29:
Sorry I was primarily assessing these 2 issues from technical perspectives. So I closed them when I was not convinced and they were no longer active. Obviously I didn't have malicious intention - just didn't want to spend time on things that I don't think worth the time but I was wrong there.

Just cut a release to address the Youtube cookie and sentry opt-inn issues: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/releases/tag/1.1.6

Cheers!

justinmbrock, 2020-04-29:
Well @tedder I felt that @foosel's comment wasn't terribly kind. I posted an issue that was relevant, and not being addressed.
Thank you @kennethjiang your plugin is impressive.

foosel, 2020-04-29:
I'm under a ton of pressure at the moment due to getting bombarded from all sides with requests of all kinds (people have or take a lot more time for their printers in these days).

I'm sorry if my curtness and fast decision making comes off as unkind, it's certainly not intended that way.

I wasnt happy with the state of things either and took this incident to introduce a new reporting type on this repository for cases like this.

justinmbrock, 2020-04-29:
You know what, I think I need to go for a drive or something, I am really stressed out. I deleted my last couple comments.

foosel, 2020-04-29:
> Just cut a release to address the Youtube cookie and sentry opt-inn issues: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/releases/tag/1.1.6

Somehow missed that, closing this since it's resolved.

closed 2020-04-29 by foosel

Mirarkitty, 2020-06-06:
Well, I did try the diplomatic way: they didn't want to resolve the issue in their own ticket... https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/180#issuecomment-619545096

But it seems like they resolved it anyway? Can someone confirm the tracker is gone?

tedder, 2020-06-06:
It was resolved and communicated on their repository under issue 51. When duplicate tickets come up it's traditional to prefer the first-filed one for communication. jneilliii's branch was a proposal and didn't reference issues as it was just a way to communicate a concept by code (sometimes easier then English). It's resolved and communicated in 51 and on this ticket.

Perhaps your objections to the plugin would be best resolved by not using it.