OctoPrint/plugins.octoprint.org #452: [Abandoned Plugin] Detailed Progress

tpmullan, 2020-04-28:
I have attempted to contact the maintainer and have not received a response in the last 6 months.  There have been many attempts by multiple people and there continues to be no response.  There are many updates that I have merged from multiple people and we would like to share with the rest of the community.

**Which Plugin? (URL to plugin page on plugin repository)**
https://github.com/dattas/OctoPrint-DetailedProgress

**Ticket on plugin's issue tracker confirming abandonment:**
https://github.com/dattas/OctoPrint-DetailedProgress/issues/40

**Is there someone open to adopting it? ("no" or @-mention user):**
I am willing to take over this plugin:
https://github.com/tpmullan/OctoPrint-DetailedProgress

foosel, 2020-04-29:
Happy to move the repo link over to your fork. Will have to put out a notice for people to update since the installed versions out there will still look to update from the original repository. I see you have already prepared a new release, is that testing and working? If so I can switch and push ASAP.

tpmullan, 2020-04-29:
I have been testing that new version for the last few months and it is stable and ready to go

jneilliii, 2020-05-02:
Glad to see my pull request finally getting merged into this plugin, glad someone was able to benefit.

tpmullan, 2020-05-02:
@jneilliii your work has been extremely helpful to me.  I really like the settings being in the web interface.

foosel, 2020-05-04:
Archive URL on repository has been pointed to your fork, and I've pushed out a notice to users that have one of the old versions installed to reinstall from the repository, that should take care of it. 

Thanks for adopting this!

closed 2020-05-04 by foosel