OctoPrint/plugins.octoprint.org #453: [Abandoned Plugin] FileManager

cconde, 2020-04-29:
**Which Plugin? (URL to plugin page on plugin repository)**

**Ticket on plugin's issue tracker confirming abandonment:**

**Is there someone open to adopting it? ("no" or @-mention user):**

foosel, 2020-04-29:
Can you please fill out the template?

jneilliii, 2020-05-02:
So not sure if this is exactly true @foosel, it appears @Salandora is still accepting pull requests since he merged one in 24 days ago to support Python3. Seems they may still be maintaining it, just not actively developing it?

Salandora, 2020-05-03:
Yeah I accepted a PR for python compatibility lately and right now update the plugins page to also state it is python 3 compatible.

On the otherhand I try a restart again as I feel like developing again and getting back into 3d-printing.

Still I would suggest to let the issue open for now just in case my comeback goes bad again...

jneilliii, 2020-05-03:
Great to see you back @Salandora, I learned so much from your plugins. Can't wait to see what you come up with next.

foosel, 2020-05-04:
Welcome back @Salandora, good to see you again! That's great news!

Have you pushed out a new release of this one yet? If so I'd say, we close this ticket, and should you get to a point again where you don't feel it anymore you can just open a new one :)

Salandora, 2020-05-04:
Last release was on the 8th April it was the python 3 compatibility release :-)
So it was before this ticket was opened.

foosel, 2020-05-05:
Closing this then. We can always reopen if there's need. Also note to self: do not just check issue tracker activity, also check releases on abandonment notification 🤦‍♀️

closed 2020-05-05 by foosel