OctoPrint/plugins.octoprint.org #466: [Abandoned Plugin] OctoPrint-Display-ETA

foosel, 2020-05-04:
**Which Plugin? (URL to plugin page on plugin repository):** https://plugins.octoprint.org/plugins/octoprint-eta/

**Ticket on plugin's issue tracker confirming abandonment:** https://github.com/pablogventura/Octoprint-ETA/issues/14

**Is there someone open to adopting it? ("no" or @-mention user):** no

AlexVerrico, 2020-05-07:
Hi @foosel 
I have copied this plugin to a new repository (https://github.com/AlexVerrico/Octoprint-Display-ETA) and uploaded it to https://pypi.org/project/OctoPrint-Display-Print-ETA/. I have also submitted pull request #467 to the plugins repository.
As well as adding python 3 support, I believe I have fixed the following issues (I can't test any of them myself, but I am sure they were all the result of a colon in a gcode command which I have removed):

- https://github.com/pablogventura/Octoprint-ETA/issues/13
- https://github.com/pablogventura/Octoprint-ETA/issues/10
- https://github.com/pablogventura/Octoprint-ETA/issues/9

Cheers,
Alex.

foosel, 2020-05-07:
See my comment on #467, there's some open points there.

closed 2020-05-18 by foosel