OctoPrint/plugins.octoprint.org #468: Action Error

jneilliii, 2020-05-08:
Not sure what happened here @foosel, but looks like the build failed when I merged a PR. Please let me know what I did wrong, if anything.

kantlivelong, 2020-05-08:
Docker build seems to run okay on my environment so am not sure why it's complaining.

kantlivelong, 2020-05-08:
https://github.com/envygeeks/jekyll-docker/issues/272

foosel, 2020-05-08:
Digging into it.

foosel, 2020-05-08:
Pinned the page build to an earlier jekyll image for now, not a permanent solution however. It seems to have something to do with the image cache at Github maybe?

jneilliii, 2020-12-09:
Is this issue still relevant @foosel?

foosel, 2021-01-25:
Probably, I haven't had a chance to test again with an unpinned setup.

foosel, 2026-09-04:
I think this can be closed, the whole build workflow was changed a while ago and we no longer use the jekyll docker

closed 2026-09-04 by foosel