OctoPrint/plugins.octoprint.org #470: [Suspicious Plugin Activity] M33 Fio

jneilliii, 2020-05-11:
**Which Plugin? (URL to plugin page on plugin repository)**

[M33 Fio](https://plugins.octoprint.org/plugins/m33fio/)

**What's the suspicious activity?**

Automatically enrolls OctoPrint instance into announcements without an opt-in option that points to a web site `https://exploitkings.com/scripts/M33 Fio.xml` [here](https://github.com/donovan6000/M33-Fio/blob/9ac4a36ebc1433294d3e28b5dfbc8be89036c2bb/octoprint_m33fio/__init__.py#L966).

**Link to ticket on the plugin's issue tracker about the issue**

https://github.com/donovan6000/M33-Fio/issues/257

donovan6000, 2020-05-12:
IIRC, OctoPrint gives you the option to disable announcements in the Announcements section of the settings. You should be able to disable M33 Fio announcements there.
[image]
If that's not satisfactory, then I'm more than happy to merge in a pull request that comments out [this line](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L978) which enables the M33 Fio announcements by default.

On a side note, I was upgrading my server last night which ended up taking down exploitkings.com for several hours. The RSS feed provided by [https://exploitkings.com/scripts/M33 Fio.xml](https://exploitkings.com/scripts/M33%20Fio.xml) is back up now.

jneilliii, 2020-05-12:
Ok, I sent a PR, but based on a couple of other issues I see in the repo, I'm not sure it's going to fix newer OctoPrint versions or not. I've been unable to get any feedback in the community forum post about it over [here](https://community.octoprint.org/t/need-to-install-older-version-to-run-m33-fio-for-micro-m3d/19290).

foosel, 2020-07-29:
@donovan6000 do you still intend to merge the PR provided by @jneilliii?

foosel, 2021-07-30:
I guess that's a no. I'm wondering, this plugin is pretty much completely dead in the water, incompatible to current versions of OctoPrint, to Python 3... shouldn't we just delist it altogether at this point?

kantlivelong, 2021-07-30:
> shouldn't we just delist it altogether at this point?

👍

jneilliii, 2021-07-30:
Agree

foosel, 2021-07-30:
Delisted.

commit fc9349f3 (2021-07-30): Delist m33fio

Closes #411
Closes #470
closed 2021-07-30 by foosel