OctoPrint/plugins.octoprint.org #471: [Abandoned Plugin] FilamentManager

DotNetRob, 2020-05-11:
**Which Plugin? (URL to plugin page on plugin repository)**
https://plugins.octoprint.org/plugins/filamentmanager/

**Ticket on plugin's issue tracker confirming abandonment:** Post the question if it was still being maintained approximately a week ago, no response from owner.   
https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/210

**Is there someone open to adopting it? ("no" or @-mention user):**

foosel, 2020-05-12:
Please give it two weeks (so one more week - life happens, Github can't take priority for everyone out there) and then ping me here again. If no reaction has happened then we can mark it as abandoned.

DotNetRob, 2020-05-20:
> Please give it two weeks (so one more week - life happens, Github can't take priority for everyone out there) and then ping me here again. If no reaction has happened then we can mark it as abandoned.

Has officially been 2 weeks since my abandonment post.

FixxCZ, 2020-06-01:
This is one of a kind plugin, it would be really nice if someone could take the ownership and keep it running. Thanks in advance.

I-like-stuff, 2020-06-03:
If no one picks this up then I found this by OllisGit who also make display layer progress and others, he said in a recent discussion that he would begin working on this after the print history plugin which seems almost finished with for it’s first version so we will just have to wait.

https://github.com/OllisGit/OctoPrint-SpoolManager

Zouljiin, 2020-06-05:
it should really be merged with 
https://github.com/OllisGit/OctoPrint-PrintJobHistory
it has great stuff and records filament usage but lacks the exact features that filamentmanager has

attacktwinkie, 2020-06-17:
I'll look into picking this up....  give me couple days to review whats here.   Ive wanted to fork this for some time and I migjt just pick this one up from here.

Kawuezel, 2020-06-29:
@attacktwinkie 
I would really like that someone is taking over the Filament Manager Plugin. I'm using it in addition to my OctoDash.

RobeeeJay, 2020-07-06:
@attacktwinkie it would be so wonderful if you could pick this up! It's broken with the new OctoPrint 1.4.1rc1. 😞

I should really learn Python one day...

cwbullet, 2020-07-10:
There might be another party interested.  My son got this working on my printers and it works with the single shared database.  Does anyone know if @attacktwinkie is planning on taking it over?

jrummy333, 2020-07-10:
who has taken this over? I would really like to add a notes section for this plugin.  Do I have to take it over just to add this or how exactly do abandon GitHub function?

OllisGit, 2020-07-15:
Hi everybody,
short advertisement: The first release candidate of the Spool Manager Plugin is out:
https://github.com/OllisGit/OctoPrint-SpoolManager

The plugin also include a note-section.

Some feature are missing for V1 (e.g. CSV Import/Export), but you can start testing and get a feeling how the plugin works. If you found an issue or have a feature request, please use the issue tracker of the repository.

BR
Olli

RobeeeJay, 2020-07-15:
This looks very nice! I would love to use it, but not being able to import my existing FilamentManager data is blocking me. I have 38 spools I'm tracking and that's a lot to manually enter all over again. 😔

OllisGit, 2020-07-22:
@RobeeeJay fyi: Its on my list for the final V1.0

OllisGit, 2020-07-28:
Hi @RobeeeJay ,
the export of the FilamentManager-Database and an CSV-Import is now implemented in RC4.
https://github.com/OllisGit/OctoPrint-SpoolManager/releases/tag/1.0.0rc4

RobeeeJay, 2020-07-28:
Thank you I will try this right now!!

buffedelic, 2020-07-29:
this is a working fork. can confirm it is runing on 1.4.0 with python 3.7

RobeeeJay, 2020-08-04:
Thank you @buffedelic! This seems to work fine and will let me upgrade Octoprint until SpoolManager is ready.

NotExpectedYet, 2020-09-10:
Is this still available to pick up?

cwbullet, 2020-09-10:
If anyone picks this up, my son has also updated it.  He is in school and does not have the time.

jneilliii, 2020-09-10:
Does @benjamin-salchow want to take over this plugin possibly? Also, @attacktwinkie mentioned it. No-one has said outright they want to take over @NotExpectedYet.

foosel, 2020-09-10:
Yep, we need someone to shout a loud and clear "here, I'll do it!" before this can proceed in any direction.

mrseeker, 2020-09-10:
Maybe @OllisGit is willing to give a shout out? It seems that his plugin does the same thing and more...

jneilliii, 2020-09-10:
Except integrate with OctoFarm...

NotExpectedYet, 2020-09-10:
> Except integrate with OctoFarm...

Wish I had the time. 

Id like to learn more about Python and its my interest to keep this plugin alive. 

I will happily maintain it and hopefully add to its feature set.

jneilliii, 2020-09-10:
That's great @NotExpectedYet will just need you to create the repo and have all the releases ready. Preferably, with the Python 3 fix already applied, etc. Once that is up and ready we can begin the transition process.

NotExpectedYet, 2020-09-10:
Sound i will get on it this weekend

RobeeeJay, 2020-09-10:
I have already moved to using SpoolManager:
https://github.com/OllisGit/OctoPrint-SpoolManager

It's in the plugins repo, you can export directly from FilamentManager to SpoolManager from within the latter's UI, and it supports more options already, plus it's actively developed.

cwbullet, 2020-09-10:
Charles knows how to find it.

NotExpectedYet, 2020-09-11:
https://github.com/NotExpectedYet/OctoPrint-FilamentManager/ 

If anyone has any commits they want to merge here's the one I forked.

OllisGit, 2020-09-12:
Unfortunately I was on summer vacation... well it was really nice.
But I missed the question, if I could pick up the Filament Manager-Plugin two days ago.
Yes, I can...maybe it is too late now.....@NotExcpectedYet or do you want to switch the responsibility to me?

NotExpectedYet, 2020-09-12:
> Unfortunately I was on summer vacation... well it was really nice.
> But I missed the question, if I could pick up the Filament Manager-Plugin two days ago.
> Yes, I can...maybe it is too late now.....@NotExcpectedYet or do you want to switch the responsibility to me?

I havent gotten around to sorting it yet with Gina and Jin but your welcome to. Id only forked the one mentioned above with python3 support. 

Your in a better positlin to keep it running than i am as i was going to be learning Python to help grow its feature set. It is fine by me if you want to take it up, and massive thanks if you do!

jneilliii, 2020-09-12:
Merged, just need to get the announcement made in OctoPrint for people to manually reinstall, @foosel?

closed 2020-09-12 by jneilliii

foosel, 2020-09-14:
> just need to get the announcement made in OctoPrint for people to manually reinstall, @foosel?

Should be live after 3849bf6938a7dea5c899ba29a74c6d7513f5abfe gets compiled. Side note: I need to build a better mechanism for this scenario right into OctoPrint, like an update config overlay that gets fetched from the repository... the notice mechanism was actually built rather for something like bugs in a specific version and thus needs individual versions, which for "anything before X" is a bit annoying. Needs to support version ranges too I guess.

PR #615 (2020-09-12): FilamentManager: Added new homepage
Hi,
I forked the repo and added the following stuff: Py3 plus several open PRs. See release notes for more details:
https://github.com/OllisGit/OctoPrint-FilamentManager/releases/tag/1.6.0
 
Now I am updating the homepage with this PR.
Please do a review, after it is approved we can close #471 

Thx, in advance
Olli