OctoPrint/plugins.octoprint.org #476: GPX Plugin: 2.6.3 update to 2.6.5 wrong download location?

wolfbiteaus, 2020-05-14:
What were you doing?

upgrading GPX Plugin: 2.6.3 update to 2.6.5

What did you expect to happen?

upgraded  GPX Plugin: 2.6.3 update to 2.6.5

What happened instead?

Updating, please wait./home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleCollecting https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gzERROR: HTTP error 404 while getting https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gzERROR: Could not install requirement https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz because of error 404 Client Error: Not Found for url: https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gzERROR: Could not install requirement https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz because of HTTP error 404 Client Error: Not Found for url: https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz for URL https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz

Could not install requirement
https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz because of HTTP error 404
*** Wrong download address?? ***
https://github.com/markwal/OctoPrint-GPX/releases/download gives 404
Manually checking location and getting the address from the file gives
https://github.com/markwal/OctoPrint-GPX/archive/2.6.5.tar.gz

(I dont know where to edit and see if it fixes)
Version of OctoPrint

OctoPrint 1.4.0 running on OctoPi 0.17.0
Operating System running OctoPrint

Octopi

I have read the FAQ.

wolfbiteaus, 2020-05-14:
[image]

foosel, 2020-05-14:
As I already told you in OctoPrint/OctoPrint#3568 you need to report this [**to the plugin author**](https://github.com/markwal/OctoPrint-GPX). The update process is completely in the hand of the plugin.

closed 2020-05-14 by foosel

wolfbiteaus, 2020-05-14:
As per your request I went look1ng for the repository and found this OctoPrint/plugins.octoprint.org.
"Please report this with the GPX plugin repository, there's absolutely nothing that I can do about that, @markwal however might be able to solve it."
Atleast this time you gave me the correct link
"As I already told you in OctoPrint/OctoPrint#3568 you need to report this to the plugin author. The update process is completely in the hand of the plugin."
Thankyou, will do.

foosel, 2020-05-14:
> Atleast this time you gave me the correct link

I'm not your personal search machine.