OctoPrint/plugins.octoprint.org #479: [Abandoned Plugin] Printer Statistics / OctoPrint-Stats

foosel, 2020-05-18:
**Which Plugin? (URL to plugin page on plugin repository)**
https://plugins.octoprint.org/plugins/stats/

**Ticket on plugin's issue tracker confirming abandonment:**
https://github.com/amsbr/OctoPrint-Stats/issues/42 - no reply after 14 days.

**Is there someone open to adopting it? ("no" or @-mention user):**
no

jmichelsen, 2020-10-04:
I'd be willing to take this up if people are still using it

cp2004, 2020-10-04:
@jmichelsen - it looks like @AlexVerrico may already be looking at taking over maintenance of the plugin. You could try getting in contact to see if you can help out.

They posted an issue in that plugin's repo asking if he could take over it.
https://github.com/amsbr/OctoPrint-Stats/issues/48

jmichelsen, 2020-10-04:
ah gotcha, thanks :)

AlexVerrico, 2020-10-04:
@jmichelsen happy to discuss sharing ownership of the plugin, might be easier to chat through gitter though: https://gitter.im/AlexVerricoChat/octoprint-stats

AlexVerrico, 2020-10-10:
G'Day all.
As I have been unable to make contact with the original maintainer, I have mirrored this repo to here: https://github.com/AlexVerrico/octoprint-stats
@jmichelsen If you want to help out with maintaining the plugin, please feel free to join the gitter chat for it https://gitter.im/AlexVerricoChat/octoprint-stats
As of now, there is no reason for listing in the plugin repo to be updated, but I will post an update here once that changes.
Cheers, Alex.

AlexVerrico, 2020-10-19:
G'Day all.
I have submitted a PR to this repo to update the plugin repo.
As such, I believe we can close this issue.
Cheers, Alex.

jneilliii, 2020-10-19:
I will keep this open for now until the merge is complete and all the notifications are created, etc.

closed 2020-10-20 by jneilliii