OctoPrint/plugins.octoprint.org #486: [Feature Request] add closed issues to github stats

jneilliii, 2020-05-23:
**What do you want to see implemented in the plugin repository?**

Now with the updated github stats on the individual plugin pages I think it would be good to include closed issues as well.

**Why, what problems would be solved by this?**

Better understanding to the end user the activeness of the plugin authors possibly.

OllisGit, 2020-05-23:
Hi @jneilliii ,

this is a little bit tricky, because the value of "closed issues" is not included in the currently used overview response (only: ```open_issues```).
Instead of using the "Issue-API" we can use the search API:
```
GET https://api.github.com/search/issues?q=repo:OllisGit/OctoPrint-DisplayLayerProgress+type:issue+state:closed&per_page=1
```

Or maybe better: Use the GraphQL-API to collect all informations with one call...I never used GraphQL, but I will take a look.

foosel, 2020-05-28:
This is now live

closed 2020-05-28 by foosel

OllisGit, 2020-05-28:
Cool, compact visualisation!