OctoPrint/plugins.octoprint.org #494: [Potential Bug] Trending Stats This Week

jneilliii, 2020-05-24:
Looking at the plugin repository it seems like the new Trending this week is showing the most recent plugins versus new installs as they are showing 0...I'm assuming this may be related to the recent github api/actions related issues (or at least I hopw).

[image]

OllisGit, 2020-05-24:
Hi,

the data for 7day-stats returns an empty-object (month-data is working):
```
Server:         nginx
Date:           Sun, 24 May 2020 06:00:34 GMT
Content-Type:   application/json
Content-Length: 2
Last-Modified:  Sun, 24 May 2020 06:00:15 GMT
Connection:     keep-alive
ETag:           "5eca0d6f-2"
Accept-Ranges:  bytes
{}
```
I think @foosel needs to take a look on it.

foosel, 2020-05-24:
Logstash crashed over a week ago on the tracking server it seems and hence no data was written to the tracking system, hence the export was empty. I just restarted it, will see if I can reconstruct some of the data from logs.

foosel, 2020-05-24:
That took longer than expected... disk ran full, had to attach a new volume, copy stuff over, run over a week of longs through the intake, and still have to get rid of some duplicates, but at least stuff is working again, including the exports and thus the week trends.

closed 2020-05-24 by foosel