OctoPrint/plugins.octoprint.org #499: [Abandoned Plugin] Filament-Reloaded

mrseeker, 2020-05-28:
**Which Plugin? (URL to plugin page on plugin repository)**
https://plugins.octoprint.org/plugins/filament_sensor_reloaded/
And most likely it's derivate:
https://plugins.octoprint.org/plugins/filamentrevolutions/ (does not have ticket system)

**Ticket on plugin's issue tracker confirming abandonment:**
https://github.com/kontakt/Octoprint-Filament-Reloaded/issues/55

Reason is that I have found a bug in the underlying code that kills octoprint, however maintainer does not seem to resolve the issue nor responds within 14 days to my request to fix this bug:
https://github.com/kontakt/Octoprint-Filament-Reloaded/issues/54

**Is there someone open to adopting it? ("no" or @-mention user):**
I created an alternative that contains a workaround, but @RomRider might be able to patch it in his build (if his plugin is not abandoned too).

RomRider, 2020-05-28:
I can patch it in my build, but I found my implementation was not ideal.

linux-paul, 2020-06-13:
Indeed meanwhile I was faced the same issue. At the end of the print Octiprint was killt. It seems to be ranomized a wrong order of function calls from callback and event handler. After  I did an further change it seems fine till yet, I think the plugin needs to be redesigned. There are parts I didn't get the intension, But I'm not a python pro

mrseeker, 2020-06-13:
I have an alternative that works @linux-paul if you need a system that works, however I still need to squash a bug where the autoreset causes another trigger.

lennvilardi, 2020-09-23:
Any updates ? or alternatives ?

ssorgatem, 2020-09-23:
I am willing to become the new maintainer.
I'm currently trying to incorporate some fixes and make it work on Python3

ssorgatem, 2020-09-23:
But I'm fairly new to Octoprint development... so... when I install my version... it doesn't show up on the plugin settings interface for some reason...

ssorgatem, 2020-09-23:
My fork (https://github.com/ssorgatem/Octoprint-Filament-Reloaded) is working well on Python2, incorporating other forks' fixes and improvements (i.e., FilamentMatrix changes and an option to prevent resuming with no filament).
Currently migrating my octorprint to Python3, let's see if it works there.

ssorgatem, 2020-09-23:
Good news: it works on Python3!

sillyfunnypedro, 2020-09-25:
@ssorgatem I am happy to help.  I have a fix for the spurious GPIO call backs that were causing the filament detector to act oddly after the first filment out event.

mrseeker, 2020-09-25:
@foosel we might have a possible maintainer

ssorgatem, 2020-09-26:
> @ssorgatem I am happy to help. I have a fix for the spurious GPIO call backs that were causing the filament detector to act oddly after the first filment out event.

Cool!
Can you share that? I'll try to add it to my fork.

ssorgatem, 2020-09-26:
I've incorporated a couple of features from other forks floating around: a navbar icon showing the sensor's status, and the option to specify whether the pin has a pullup or pulldown resistor.

sillyfunnypedro, 2020-09-26:
Here you go.   The problem was that the GPIO call back is sometimes called
when there is no change in any of the pins, i just added code to check the
state of the pin to see what was happening.  Ive been using this for about
a year now and each time i have a filament out and i replace it the system
behaves as it should.  Someone else also used this patch.

ssorgatem, 2020-09-27:
I don't see the patch or any code?

sillyfunnypedro, 2020-09-27:
i attached a zip file containing the whole project.  here is a copy of the
file

ssorgatem, 2020-09-27:
Oh, you are replying by email?
I don't think the attachments are visible in the comment thread. So I can't see any attachment (neither can I on my email notifications).

So you'll probablky have to either send it to me separately (my email address is my github username @gmail.com), or post it somewhere else and send the link here.

sillyfunnypedro, 2020-09-27:
https://www.dropbox.com/s/x5twi87f21wc32j/filamentforpull.zip?dl=0

Got it,... sorry just waking up here.

can you see this?

ssorgatem, 2020-09-27:
yes, thank you, I just downloaded it now

ssorgatem, 2020-09-27:
I still have to review your code, but I've made a PR for what I already have
https://github.com/OctoPrint/plugins.octoprint.org/pull/638

ssorgatem, 2020-09-27:
@sillyfunnypedro  I merged your patch into a branch: https://github.com/ssorgatem/Octoprint-Filament-Reloaded/tree/sillyfunnypedro

Can you test if it works as it was intended? (it conflicted with some of the changes I've been adding...)
I'm currently in the middle of a long print...

sillyfunnypedro, 2020-09-27:
I will test this evening.

jneilliii, 2020-09-27:
Adoption PR has been merged, closing this issue now. Thanks for picking this up and keeping the plugin alive.

closed 2020-09-27 by jneilliii

sillyfunnypedro, 2020-09-28:
I pulled the branch sillyfunnypedro and i installed it on my octoprint
server.
i started a print and then 3 times removed the filament from the sensor and
had the printer pause correctly.  when i reinserted the filament and
restarted it worked fine.

the spurious GPIO issue that i fixed did not manifest.

ssorgatem, 2020-09-28:
A new release is out including your fix!
https://github.com/ssorgatem/Octoprint-Filament-Reloaded/releases/tag/1.3.1

mrseeker, 2020-09-28:
To prevent a lot of messages non-related to my issue (the plugin being abandoned), I recommend creating a new ticket at the new maintainers GitHub repository if you keep having issues with the plugin.