OctoPrint/plugins.octoprint.org #502: [Listing Error] Detailed Progress - This plug-in is Python 3 compatible

benjamin-salchow, 2020-05-30:
**Which Plugin? (URL to plugin page on plugin repository)**

Detailed Progress

https://plugins.octoprint.org/plugins/detailedprogress/

https://github.com/tpmullan/OctoPrint-DetailedProgress

**What's wrong with the plugin *listing*?**

The Plug-in is compatible with Python 3 , but it's flagged as non compatible with Python 3 (< 3 ).

https://github.com/tpmullan/OctoPrint-DetailedProgress/pull/3

This is merged and available in master.

https://github.com/tpmullan/OctoPrint-DetailedProgress/blob/master/octoprint_detailedprogress/__init__.py#L202
<- line of selected compatibly
```__plugin_pythoncompat__ = ">=2.7,<4"```

Thank you very much and have a great day :-)

jneilliii, 2020-05-30:
@benjamin-salchow it's the plugin's author who is responsible for submitting a PR to update this. Please let @tpmullan know he missed a step documented [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html#telling-octoprint-your-plugin-is-python-3-ready) just after the note.

closed 2020-05-30 by jneilliii

tpmullan, 2020-06-01:
@jneilliii I see you added the compatibility here: https://github.com/OctoPrint/plugins.octoprint.org/commit/cc8c83febd099ea303b718b38f5e8ea033c0b50a#diff-58759881c64ae2d5a42e25c292d40f30

Is there anything else I need to do to support python 3?

jneilliii, 2020-06-01:
@tpmullan that change was for the plugin displayprogress, not detailedprogress. You need to add those same lines to detailedprogress.md and submit a PR so the plugin repository knows your plugin is Python 3 compatible.