OctoPrint/plugins.octoprint.org #521: Link Checker

jneilliii, 2020-06-08:
I just noticed a couple of my plugins that were recently merged were missing screenshots, and after looking into it, issue seemed to be related to invalid links, both in the featured image section and within the detailed description at the bottom. I think this is a result of switching from Travis to Github Actions as the Travis build would error if links were invalid. I remember the checks would not pass before if the links didn't work.

foosel, 2020-06-12:
There's now a link checker as part of the validation steps.

foosel, 2020-06-12:
Turns out the link check caused the build to run into rate limiting issues, so I had to disable it again. Might have to use a different check, ideally with result caching between checked pages (the one right now checks links that are included in the footer for example again and again and again)

foosel, 2026-01-19:
That would be lovely. The first run will probably run into the same rate-limiting issue, but maybe the cache solves that by detecting the situation and handling it better...

jacopotediosi, 2026-01-20:
I tried lychee.

Regarding the link status caching feature, it works well, but I don't think we actually need it: instead of adding a step to the GitHub Pages deploy action, I believe it would be more useful to create a separate action that runs less frequently, perhaps weekly, and that doesn't block the deploy but still report invalid links by opening issues or logging to a discord channel.

Regarding rate limits, lychee allows specifying a GitHub token to avoid them for GitHub links, but this repository contains so many that it happens anyway.

I think it's better to wait for this feature to be released (merged but not yet released), which should solve the problem definitively:
- https://github.com/lycheeverse/lychee/issues/1605