OctoPrint/plugins.octoprint.org #564: [Suspicious Plugin Activity] OctoPrint-WebcamStreamer

GitHubEmploy, 2020-07-28:
https://github.com/jneilliii/OctoPrint-WebcamIframe/archive/master.zip

It makes it printer randomly restart and it also sometimes takes tracking info from my network

foosel, 2020-07-29:
Random restarts sounds more like an issue with your printer or your power supply. Even more so as the plugin in question has next to no backend parts. What makes you think it's this plugin?

Taking a look at the source I also don't see anything hinting at tracking (and I would also be severely surprised about finding anything given the author of the plugin). Could you please clarify what you are seeing that makes you think something is "taking tracking info from your network"?

Also cc @jneilliii as it's one of his plugins.

jneilliii, 2020-07-29:
In no way does my plugin do any of that. Question though, title says one plugin, link says another, which plugin is it?

closed 2020-08-05 by foosel