OctoPrint/plugins.octoprint.org #572: [Abandoned Plugin] Full-featured Slicer

mrseeker, 2020-08-05:
**Which Plugin? (URL to plugin page on plugin repository)**
https://plugins.octoprint.org/plugins/slicer/

**Ticket on plugin's issue tracker confirming abandonment:**
https://github.com/kennethjiang/OctoPrint-Slicer <- see readme.md

**Is there someone open to adopting it? ("no" or @-mention user):**
Unsure

foosel, 2020-08-05:
Thanks, it's marked now.

adamoutler, 2020-09-23:
This is important to me because I use mobile and Chromebook for printing.    There is no good way to use Cura on Mobile, or Chromebook.  Also it is much cleaner to have everything in a single place like Octoprint.  My appreciations will go to anyone who maintains this plugin.

atxbyea, 2020-09-28:
> This is important to me because I use mobile and Chromebook for printing. There is no good way to use Cura on Mobile, or Chromebook. Also it is much cleaner to have everything in a single place like Octoprint. My appreciations will go to anyone who maintains this plugin.

I know this doesn't help, I wish too I had a fast and easy slicer in octoprint, since I too am a lot of mobile and chromebook, but I installed my chromebook with galliumos (which gave me a lot more freedom at work to run other applications) which might be an option for you, or not, just throwing it out there (you can also dualboot gallium and chromeOS).
Other option that I am currently exploring is doing the slicing via CLI since a lot of slicers support this, I haven't had time to tinker with this yet, hope it atleast is of some use to you until we (maybe) get a slicer in octoprint running again.

jneilliii, 2020-09-28:
Have you guys tried the gridspace plugin yet?

https://plugins.octoprint.org/plugins/gridspace/

yenalsekim, 2021-01-02:
This project needs more than I could offer as one person but I'd love to offer my help if somebody took the lead. 
I can read code a bit, test an app a quite a bit more and document almost as well as I make coffee... and I make a mean cup.  That's not to say that I write mean documents, I'm merely sayi....  
You know what I'm saying.

eyal0, 2021-01-16:
@jneilliii @yenalsekim @adamoutler  I used to work a lot on my own fork of OctoPrint-Slicer.  I even pushed some PRs a few years ago.

Right now OctoPrint-Slicer at master is broken because of two things:

1. python3 support is missing
2. A jQuery update in November from OctoPrint broke it.

I have sent two PRs on OctoPrint-Slicer that will repair it.  The repo says that it has a lot of installs so maybe it's worth repairing?  I have no idea how so many people are using it right now given that the GUI is busted but we might as well repair it.

If someone wants to try my fork right now, install this:

https://github.com/eyal0/OctoPrint-Slicer/archive/master.zip

jneilliii, 2021-01-16:
Are you volunteering to adopt maintenance of this plugin @eyal0? If so the process is pretty easy. You just have to submit a PR that changes all the relative install URLs etc.

yenalsekim, 2021-01-16:
Hi! 
Thanks for the email!
Yes, if the project is popular and it adds value to OctoP I'd love to help. 
It needs a slicer.  This one has already got it's foot in the door. 
Not a programmer but I get it. 
I can test and write the crap out of anything.
If that would help the project then ya, I'm in. 

Mike

yenalsekim, 2021-01-16:
Hi Guys,
I'm Mike. Thanks for pulling me into the loop. 

I'll try the fork tonight and give it a go.

eyal0, 2021-01-16:
My fork has some new features which I think are cool but might also slow down the slicer.  Or it might run just as fast or faster. 
 I don't know.  So please try both:

More features but maybe runs slower: https://github.com/eyal0/OctoPrint-Slicer/archive/master.zip
The original one: https://github.com/eyal0/OctoPrint-Slicer/archive/old.zip

It's been a while since I worked on my fork so I don't remember all the feature differences but I think that it's like this:

* There's a chop button that will let you cut and object in two.  Maybe this one exists in both versions, I don't remember.
* Perhaps more slicer settings to support slic3r, I think that the perimeters parameter is available.
* My version has collision detection so your model should turn red if it occupies the same space as another model or if it sticks out over the edge of the printable area.
* The "tilt" algorithm to lay your object flat if you rotate it is much more advanced.  (It does a sort of physics simulation so that the object will fall until the center of gravity is supported.)
* There is an advanced arrange button that will try very hard to get all your object to fit on to the build surface.  (It does a bin packing algorithm.)

Even without the advanced features it runs kinda slow on my computer.  Might just be my computer, though.  Let me know how they run for you, let me know which one seems more performant.  Maybe they seem the same anyway?

Thanks!

yenalsekim, 2021-01-17:
I like how we keep our options open!

eyal0, 2021-01-17:
> Are you volunteering to adopt maintenance of this plugin @eyal0? If so the process is pretty easy. You just have to submit a PR that changes all the relative install URLs etc.

Thanks @jneilliii I will.  While you're waiting on that, I got 6 tiny PRs waiting that are like one-liners in OctoPrint.  :-)

jneilliii, 2021-01-17:
Yeah, those OctoPrint core PRs are not reviewed/merged by me. That's all Gina. She typically looks over those at the beginning of the week depending on what's going on.

jneilliii, 2021-01-19:
@kennethjiang has given @eyal0 direct access to the repo and it is now officially adopted.

closed 2021-01-19 by jneilliii