OctoPrint/plugins.octoprint.org #574: [Listing Error] DisplayLayerProgress

deeznuts54229, 2020-08-07:
Every time I login it tells me that the display
 "Cant get stats from
DisplayLayerProgress."   
I have a prusa i3 mk3s so I have no clue why it does not work. 

**Which Plugin? (URL to plugin page on plugin repository)**

**What's wrong with the plugin *listing*?**

foosel, 2020-08-07:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin **repository**, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
>
>   * missing or broken homepage, source or download URLs
>   * broken images
>   * incorrect compatibility information
>   * ... in short, anything related to the data on plugins.octoprint.org
>
> What you should not report here:
>
>   * Any kinds of issues with the plugins themselves - report those
>     to the plugin author instead!

Thanks :)

closed 2020-08-07 by foosel

OllisGit, 2020-08-07:
Hi @deeznuts54229 ,
the issues is closed, but if you report the issue in my issue tracker, please also add a screenshot.
I can't imagine where this error could appear. There is no "stats" in the plugin.

OllisGit, 2020-08-07:
Hi @deeznuts54229 ,
I search for "get stats" in my dev environment....and yeah I found it in the Dashboard plugin.

Please raise a ticket at https://github.com/StefanCohen/OctoPrint-Dashboard/issues

Btw. did your read the full message:
```
Cant get stats from DisplayLayerProgress.  
This plugin is required and provides GCode parsing for Fan Speed, Layer/Height info, 
Layer Durations and Average layer time. Is it installed, enabled and on the latest version?
```