OctoPrint/plugins.octoprint.org #599: [Abandoned Plugin] OctoPrint Klipper Plugin

AliceGrey, 2020-08-29:
**Which Plugin? (URL to plugin page on plugin repository)**
https://github.com/mmone/OctoprintKlipperPlugin

**Ticket on plugin's issue tracker confirming abandonment:**
Last release was 2 years ago
https://github.com/mmone/OctoprintKlipperPlugin/releases/tag/0.2.5

**Is there someone open to adopting it? ("no" or @-mention user):**
I am open to adopting it

jneilliii, 2020-08-29:
@AliceGrey you linked to the releases, which is not necessarily an indication of abandonment, although based on the open issue count and pull requests it does in fact appear to be abandoned.

@mmone it appears there are several pull requests against your repo that will fix several issues, etc. Are you no longer actively developing this plugin? If you are not or we don't hear back from you within 14 days we will transition this plugin to @AliceGrey's fork and update the official plugin repo accordingly.

@AliceGrey relative to that transition, I'm curious if you merged in all those Pull Requests that fix other issues in your fork?

AliceGrey, 2020-08-29:
@jneilliii 
I pulled in the PR's except for https://github.com/mmone/OctoprintKlipperPlugin/pull/64
I have yet to review it fully. The other two PR's resolved the issues mentioned in the original PR's and have been merged. I also have confirmed in a local dev environment that the plugin runs on python 3 after some minor tweaks.

jneilliii, 2020-08-29:
Perfect, if we don't hear back from @mmone we'll make the switch.

jneilliii, 2020-09-06:
Just one final follow-up @mmone, if we don't hear back from you in the next 7 days we will transition this official listing to @AliceGrey's maintained fork.

jneilliii, 2020-09-12:
@AliceGrey ok, all I need from you now that we've waited for the proper number of days is a PR for the plugin repo md page with corrected URLs to your repo, etc. Once I've gotten that and reviewed we'll submit the announcement for people to manually reinstall to get them back in line for future software update checks.

cowboy3d, 2020-09-12:
Just saw this activity... after having created a pull request against my repository. @AliceGrey: Have a look at https://github.com/cowboy3d/OctoprintKlipper3Plugin if you're interested to see what I have changed to make it work with python 3.
Thank you for taking this over and good luck!

jneilliii, 2020-09-15:
PR merged @AliceGrey, thanks for adopting this plugin and keeping it alive. @foosel ready for a change notice.

jneilliii, 2020-09-15:
Disregard Gina, I just realized I can copy the last one you did. I'm pretty sure I understand how it works. Will submit a PR for your review just to make sure.

closed 2020-09-15 by jneilliii

foosel, 2020-09-15:
PR review and merged, you did do it perfectly :)

PR #616 (2020-09-15): Update OctoKlipper Plugin Repo
Update OctoKlipper Plugin Repo to new fork. Resolves #599