OctoPrint/plugins.octoprint.org #608: [Abandoned Plugin] OctoPrint-Cost

cp2004, 2020-09-07:
**Which Plugin? (URL to plugin page on plugin repository)**
https://github.com/jasiek/OctoPrint-Cost 
https://plugins.octoprint.org/plugins/cost/
**Ticket on plugin's issue tracker confirming abandonment:**
Top of readme.md, mentions that https://github.com/malnvenshorn/OctoPrint-CostEstimation is being maintained, instead of this one
https://github.com/jasiek/OctoPrint-Cost
**Is there someone open to adopting it? ("no" or @-mention user):**
already been an alternative registered?

foosel, 2020-09-22:
Sadly it looks like the listed alternative is also abandoned: #623

OllisGit, 2020-09-22:
Hi, 
I created a PR for the "OctoPrint-CostEstimation" plugin. 
I didn't see any difference between both plugins. IMHO we should remove this one.

atomicalsoftwares, 2020-09-29:
How do I ask to maintain the repository?

jneilliii, 2020-09-29:
You just did. Do you want to take over this plugin as the maintainer?

pete-oconnell, 2020-09-30:
I'd like to look at maintaining the repo

JTensai, 2021-01-26:
This is still marked as an "Abandoned plugin" any progress on changing that?

jneilliii, 2021-01-26:
No one has stepped up to the plate and proven that they will maintain it or submit a PR to adopt the plugin, so not really sure where this stands.

awenelo, 2021-01-26:
It looks like the listed alternative, OctoPrint-CostEstimation is being maintained, by @OllisGit: https://github.com/OllisGit/OctoPrint-CostEstimation

OllisGit, 2021-01-28:
As already mentioned, let's drop it ;-) IMHO Because we don't need two plugins with the exact same feature set.