OctoPrint/plugins.octoprint.org #619: [Abandoned Plugin] M73 ETA Override

adelyser, 2020-09-18:
**Which Plugin? (URL to plugin page on plugin repository)**
https://github.com/sysadminsh/OctoPrint-M73ETAOverride/issues/18

**Ticket on plugin's issue tracker confirming abandonment:**
[Feature Request] Python 3 Compatibility #18

**Is there someone open to adopting it? ("no" or @-mention user):**
no

ShanDestromp, 2020-09-18:
[Here](https://github.com/ShanDestromp/OctoPrint-M73ETAOverride); includes the Python3 fix and I even went ahead and added [#17](https://github.com/sysadminsh/OctoPrint-M73ETAOverride/pull/17)  This should be considered temporary, I only know enough python to be dangerous and active development is beyond my talents.  Basically someone should take this over ASAP before I hurt myself.

That said I installed via zip and everything *seems* to be working as it should; although plugin manager didn't enable it by default on install (might have been because I'd disabled the old one and it saved that state).

cp2004, 2020-09-19:
There is also [OctoPrint Prusa M73 Override](https://github.com/kanocz/octopi_eta_override), [repository listing](https://plugins.octoprint.org/plugins/prusaetaoverride/) that I updated (Py 3) & tested a while back, it looks like it does the same thing? If so, there may not be a need to find a new maintainer. The author (kanocz) on that repository is responsive to PRs but not actively developing the plugin.

Moorelife, 2020-09-22:
> **Is there someone open to adopting it? ("no" or @-mention user):**
> @Moorelife (a.k.a. I) am into delving into  this plugin. I recently got into 3D printing and using OctoPrint, finding both very interesting. As a Test Automation engineer I have sufficient knowledge of both Python 3 and 2. Feel free to speak up if you think I shouldn't take this task in hand.... :-)

Moorelife, 2020-09-23:
> [Here](https://github.com/ShanDestromp/OctoPrint-M73ETAOverride); includes the Python3 fix and I even went ahead and added [#17](https://github.com/sysadminsh/OctoPrint-M73ETAOverride/pull/17) This [...]

@ShanDestromp consider this taken over.  :-) The adopted code can be found [here](https://github.com/Moorelife/OctoPrint-M73ETAOverride). Thanks for doing the quick fix, that will give me some time to delve into the Octoprint plugin tutorial and the code, before actually getting into de active development part.

ShanDestromp, 2020-09-23:
@foosel @adelyser That's one you can scratch off the abandoned list 👍 

@Moorelife Appreciate the take up.  I just happened to be looking at the repo and saw the work was already done.

cp2004, 2020-09-23:
@Moorelife - Does this mean you will be able to take over the repository listing straight away (ie. new users install your plugins, old ones get told to migrate), or you would like a little more time?

When you want to officially take over, feel free to fork this repository and edit the plugin's listing to point to your fork, and anything else you want changed. Then send in a PR back here and it can be official!

Moorelife, 2020-09-23:
Hi,I have already forked the project, will be driving into the project over the next few days...Any hints and tips you can give me during that time are appreciated, I aim to be up and running within 2 weeks.Best Regards,André

Moorelife, 2020-09-23:
Hmm, although the plugin tutorial could do with a few small updates, it was enough to see plugin development is quite straightforward. Now delving into the code of the M73ETA plugin itself...

gdombiak, 2020-09-25:
I was going to offer taking it but happy to see @Moorelife already took it! Thanks André

richardsondev, 2020-12-03:
Hey there, has there been any update to this?
Best regards

Moorelife, 2020-12-03:
No Sorry Billy,

Since a lot of stuff happened in the last few weeks in my life, I haven't been able to pick this up yet, and am even having second thoughts about it because it seems I've spread myself too thin....

If anyone else feels up to the task, I welcome their effort.

Best Regards,

André Slabber

gdombiak, 2020-12-03:
I would be happy to take ownership on this plugin if no one took it already!

Gaston

Moorelife, 2020-12-03:
Sure Gaston,

Thanks a lot for stepping in. Sources are at <https://github.com/OctoPrint/plugins.octoprint.org>

Best Regards,

André

gdombiak, 2020-12-04:
Hi @Moorelife, do you happen to know the process for taking ownership? I guess I will fork the project, update existing entry in https://github.com/OctoPrint/plugins.octoprint.org? How do I mark it as no longer abandoned?

Thanks,
Gaston

jneilliii, 2020-12-04:
We can handle that part once you have the fork ready and have updated the update check hook to point to your repo and release a new version on it. Make sure to update the version number in setup.py before creating the release.

jneilliii, 2020-12-04:
but yes, you'll need to create a PR against this plugin's repo with author/authors updated and updated URLs to your fork once you have the release ready. It's similar to what you would do if you were registering a new plugin as described [here](https://plugins.octoprint.org/help/registering/). Please just make sure to do it from a separate branch on your fork of plugins.octoprint.org.

gdombiak, 2020-12-05:
Thanks @jneilliii for the info. Will work on this during the weekend. Trying to finish Siri control of Enclosure plugin

gdombiak, 2020-12-05:
All done. Made the plugin compatible with Python 3. Updated all information plus improved description and setup. [PR](https://github.com/OctoPrint/plugins.octoprint.org/pull/699) sent.

PR #705 (2020-12-08): m73etaoverride adoption
Add necessary notices for plugin adoptions. finalizes #699, closes #619
commit 9f93340f (2020-12-08): m73etaoverride adoption notices (#705)

Add necessary notices for plugin adoptions. finalizes #699, closes #619
closed 2020-12-08 by jneilliii