OctoPrint/plugins.octoprint.org #622: [Abandoned Plugin] continuous print

mrseeker, 2020-09-20:
**Which Plugin? (URL to plugin page on plugin repository)**
https://plugins.octoprint.org/plugins/continuousprint/

**Ticket on plugin's issue tracker confirming abandonment:**
https://github.com/nukeem/continuousprint/issues/20

**Is there someone open to adopting it? ("no" or @-mention user):**
Unknown

jneilliii, 2020-10-30:
@Zinc-OS is that you volunteering to adopt this plugin? If so please create your fork in your github and adjust the software update hook stuff in the plugin. Make sure to bump the version in setup.py and create a release that matches on your repo when these changes are completed. Once all that is done submit a PR here with the necessary changes to URLs in the plugin listing. We only ask that you keep the original plugin author in some way. There are a couple of examples already, like Charlie Powell, previously Anderson Silva. We will review and then update and make the notices as necessary.

Zinc-OS, 2020-10-30:
Yes. I would like to adopt it.

jneilliii, 2020-10-30:
Ok, once you've done the steps above we'll be good to proceed with migrating the plugin listing to your fork.

closed 2020-10-31 by jneilliii