OctoPrint/plugins.octoprint.org #623: [Abandoned Plugin] OctoPrint-CostEstimation

ikoutone, 2020-09-20:
**Which Plugin? (URL to plugin page on plugin repository)**
https://github.com/malnvenshorn/OctoPrint-CostEstimation

**Ticket on plugin's issue tracker confirming abandonment:**
https://github.com/malnvenshorn/OctoPrint-CostEstimation/issues/29

**Is there someone open to adopting it? ("no" or @-mention user):**
No

jneilliii, 2020-09-21:
@benjamin-salchow since you have a python 3 compatible fork, do you want to take over maintaining this plugin?

RedCoder09, 2020-09-22:
OH NO!! I would take over, but I know nothing about coding so :/

OllisGit, 2020-09-22:
Hi everybody,

I can take over the plugin responsibility.
I am just preparing the MR for the Plugin-Repo and also looking through the current open merge requests.

jneilliii, 2020-09-23:
Pugin adopted, changes merged.

closed 2020-09-23 by jneilliii