OctoPrint/plugins.octoprint.org #628: [Abandoned Plugin] OctoPrint-Dashboard

StefanCohen, 2020-09-21:
**Which Plugin? (URL to plugin page on plugin repository)**
https://plugins.octoprint.org/plugins/dashboard/

**Ticket on plugin's issue tracker confirming abandonment:**
https://github.com/StefanCohen/OctoPrint-Dashboard/issues/203

**Is there someone open to adopting it? ("no" or @-mention user):**
no

ochm, 2020-09-22:
Sad :(

ZeekWorld, 2020-09-22:
One of the best dahsboards out there, bummer.

StefanCohen, 2020-09-22:
Thanks. I really wish I had the same amount of time now as I had when I started the project. Three kids, a full renovation of a house located four hours away on top of a demanding full-time job doesn't leave much time for hobbies.

Willmac16, 2020-09-22:
I mentioned on the other issue, but I am open to taking over the plugin.

sonny-c, 2020-09-24:
Will the plugin continue to work without being updated or maintained?

j7126, 2020-09-24:
> Will the plugin continue to work without being updated or maintained?

it should yes

slim0287, 2020-09-24:
Seems like at least two people here have volunteered to take over it (@j7126  and @Willmac16 ).  How do we know if one formally gets the job?

foosel, 2020-09-24:
Y'all have to agree on something first, we don't decide this, it's y'all ;) Partnering in such cases can be a good option too.

j7126, 2020-09-24:
What is the process for actually transferring the plugin?

I would be happy to take over the plugin myself, or if @Willmac16 wants we could partner. 

I guess the final decision would be up to @StefanCohen?

foosel, 2020-09-24:
Since here the author is still reachable and involved in the adoption process, there a more options than we usually have. The easiest for everyone really would be if @StefanCohen simply gave commit access to his repository to the new maintainer(s). That way the adoption can be handled completely by him without any involvement or changes needed by the plugin repository here.

Alternatively the plugin's repo could be moved/transfered in ownership or forked. I *think* in the former case there would be minimal involvement required from use here, the only thing needed would be a change in the plugin listing. Plugin updates *should* be handled transparently by Github in such a case since I think they'll setup the needed redirects automatically (at least that's how it happened when I moved OctoPrint itself from foosel/OctoPrint to OctoPrint/OctoPrint). 

In case of a fork we'll need to push out a notification and an update overlay to people who have already a version installed so everyone knows where to get the plugin updates from now on. I'd prefer to avoid this as it's extra work for the plugin repository maintainers and ALSO the least comfortable for all existing users out there. This options is what we usually have to resort to when the maintainer no longer responds though.

j7126, 2020-09-24:
Checking the GitHub docs for transferring repo ownership it says that it will automatically create all the necessary redirects so that seems like it would be the way to go.

Willmac16, 2020-09-24:
I'd be happy to partner!

StefanCohen, 2020-09-24:
Great. Let's take the transition discussion over in this ticket: https://github.com/StefanCohen/OctoPrint-Dashboard/issues/203 and make it official here when we agree.

vtozato, 2020-09-25:
An excellent plugin, made it so easy ... I was sad.

simonpucher, 2020-09-26:
An awesome plugin! Maybe someone will be able to continue with this project and @StefanCohen will rejoin when his "other projects" (🏡👨‍👩‍👦🗄) are finished.

jerm, 2020-09-26:
@StefanCohen Really appreciate all your efforts this plugin. Extra thanks for coordinating a graceful handoff -- it's so easy to let side projects languish when life gets in the way. 🎉

StefanCohen, 2020-09-27:
Thank you all for the support. I have now transferred the plugin ownership to @j7126 and @Willmac16 who will collaborate on maintaining the Dashboard going forward. This ticket can be closed and @j7126 will prepare a pull request to the plugin repository with the changes.

j7126, 2020-09-27:
Myself and @Willmac16 will now be maintaining the plugin. I have submitted the pull request with the required changes to the plugin page #637.

j7126, 2020-09-28:
This issue can probably be closed now

closed 2020-09-28 by StefanCohen