OctoPrint/plugins.octoprint.org #646: [Abandoned Plugin] AutomaticShutdown

madjack911, 2020-10-02:
**Which Plugin? (URL to plugin page on plugin repository)**

https://plugins.octoprint.org/plugins/...?

**Ticket on plugin's issue tracker confirming abandonment:**

**Is there someone open to adopting it? ("no" or @-mention user):**

madjack911, 2020-10-02:
no support to python 3

foosel, 2020-10-02:
Can you please fill out the template?

foosel, 2020-10-09:
Confirmed with @kantlivelong

oechslein, 2020-10-11:
There is a fork of osiu97 (https://github.com/osiu97/OctoPrint-AutomaticShutdown) that states it works with python3.

madjack911, 2020-10-11:
thanks oechslein  for the update to python 3 .work fine. thank you very much

madjack911, 2020-10-11:
and thank you foosel for your help

jneilliii, 2023-01-07:
I don't think this is valid anymore? Looks like @kantlivelong merged my patch PR for python 3 compatibility.

n0sr3v, 2023-08-01:
As stated by @jneilliii plugin has been updated [2023-01-18](https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/releases), so i would not consider this as abandoned any more.
Just made a push request to remove the "abandoned"-flag, it's time to get rid of the annoying notification!

closed 2023-08-01 by jneilliii